Fr4nkFletcher/ESP32-Marauder-Cheap-Yellow-Display #1: capacitive touch?

eried, 2024-02-04:
hi, does this work on the version with capacitive touch?

v7xDm1r, 2024-02-04:
> hi, does this work on the version with capacitive touch?

Are they not all capacitive? Yes it works with capacitive with the New espi library.

Fr4nkFletcher, 2024-02-04:
The only two boards I've tested on (so far) are mentioned in the readme. One with type-c&microusb vs micro only

eried, 2024-02-04:
oh ok, there is a [capacitive ](https://www.aliexpress.com/item/1005005865107357.html?spm=a2g0o.order_list.order_list_main.5.11c01802JpXO49) touch one, I will order and test :D

closed 2024-02-04 by eried

v7xDm1r, 2024-02-05:
> oh ok, there is a [capacitive ](https://www.aliexpress.com/item/1005005865107357.html?spm=a2g0o.order_list.order_list_main.5.11c01802JpXO49) touch one, I will order and test :D

Usually you need 4 files. But I may be wrong. You'll have to wait for a replay from creator.

Fr4nkFletcher, 2025-03-20:
@eried finally working for most of the capacitive touch boards

Fr4nkFletcher, 2025-03-31:
@eried which board?

eried, 2025-03-31:
> [@eried](https://github.com/eried) which board?

https://www.aliexpress.com/item/1005005865107357.html

Fr4nkFletcher, 2025-03-31:
> https://www.aliexpress.com/item/1005005865107357.html

Works now