Fr4nkFletcher/ESP32-Marauder-Cheap-Yellow-Display #2: Bin file doesn't seem to work

ashdavely, 2024-02-05:
Flashed bin file using the instructions for esptool. Flash completes but just get blank screen after reset.

v7xDm1r, 2024-02-05:
> Flashed bin file using the instructions for esptool. Flash completes but just get blank screen after reset.

Did you use all 4 files needed and make sure they're in the proper location?

ashdavely, 2024-02-05:
I followed the instructions for the webflasher. It doesn't mention other files. Just 2 bin files, the second to try if there is color issues. What did I miss?

witnessmenow, 2024-02-05:
Did you update the address of the flash address? 0x1000 is the default, 0x10000 is what it says on the readme

espino1, 2024-02-05:
I have flashed the .bin file to the correct address and the display doesnt work.   the log just cycles when I view it.

JediMongoose, 2024-02-05:
Same problem here. Changed the address to 0x10000 but only get black screen.
Using a serial terminal, I get this repeated over and over:
`
rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
ets Jul 29 2019 12:21:46
`

What are the 3 other files @v7xDm1r ?

ashdavely, 2024-02-05:
I changed the flash address. Tried to erase beforehand also. I've noticed now that if I flash one of the CYD test projects the screen doesn't work as well.

Fr4nkFletcher, 2024-02-05:
Working on this now. Please compile from source and let me know

JediMongoose, 2024-02-05:
Used your web flasher mentioned on the home page - and now working fine - thank you! :+1:

ctjameson, 2024-02-06:
@Fr4nkFletcher the web flasher works great. Could we get an inverted option?

Fr4nkFletcher, 2024-02-06:
> @Fr4nkFletcher the web flasher works great. Could we get an inverted option?

All set

ctjameson, 2024-02-06:
My dude! Can confirm working on my USB-C CYD recently acquired from Ali. 

Would it be possible to just add the inverted flag as a setting in the menu instead of having to maintain two different files/flasher sites?

Fr4nkFletcher, 2024-02-06:
> My dude! Can confirm working on my USB-C CYD recently acquired from Ali.
> 
> Would it be possible to just add the inverted flag as a setting in the menu instead of having to maintain two different files/flasher sites?

Like this? https://fr4nkfletcher.github.io/ESP32-Marauder-Cheap-Yellow-Display/flash0.html

ctjameson, 2024-02-06:
> Like this? https://fr4nkfletcher.github.io/ESP32-Marauder-Cheap-Yellow-Display/flash0.html

That works, but I meant inside of Marauder after install. This way you wouldn't need two different bin files to keep up with. Wasn't sure if Marauder is given that level of configuration post boot or not.

closed 2024-02-06 by Fr4nkFletcher