Fr4nkFletcher/ESP32-Marauder-Cheap-Yellow-Display #5: Input issue after using 0.13.7 flash tool

NoxiousKarn, 2024-02-07:
It seems that when entering a rotated menu the screen rotates -90 degrees top counter clockwise but the touch controls rotate 90 degrees so top of unit rotates clockwise... I had to roll back to 0.13.5 where the problem does not exist.
https://youtu.be/UI-ROiRZYok

ggaljoen, 2024-02-07:
Tip to solve this is in this calculation of the orientation:
lcd.setRotation(lcd.getRotation() ^ 1)

Fr4nkFletcher, 2024-02-07:
Gotcha. Will try to find time this evening to fix and reupload

Fr4nkFletcher, 2024-02-07:
> Pull request made

Thanks man. Will redo the bin soon, just working right now

Fr4nkFletcher, 2024-02-08:
> It seems that when entering a rotated menu the screen rotates -90 degrees top counter clockwise but the touch controls rotate 90 degrees so top of unit rotates clockwise... I had to roll back to 0.13.5 where the problem does not exist. https://youtu.be/UI-ROiRZYok

Good to go, let me know

NoxiousKarn, 2024-02-08:
works great now Thanks!

closed 2024-02-08 by NoxiousKarn