Fr4nkFletcher/ESP32-Marauder-Cheap-Yellow-Display #10: Evil Portal Question

ChromeBlue, 2024-02-22:
Where do I place the portal html files?  I'm assuming on the SD but what folder/s    Also ap.config.txt and index.html?

Fr4nkFletcher, 2024-02-23:
> Where do I place the portal html files? I'm assuming on the SD but what folder/s Also ap.config.txt and index.html?

WiFi -> Sniffers -> Scan APs -> WiFI General -> Add SSIDS (manually) or Select APs -> Select EP HTML File.

Your fake page .html pages will go in the root directory of the SD card.

closed 2024-02-23 by Fr4nkFletcher

ChromeBlue, 2024-02-25:
> <Fr4nkFletcher/ESP32-Marauder-Cheap-Yellow-Display/issues/10/1962141738@
> github.com>

Thank you for your response.  I had figured it out but I thought I'd leave
the question posted because I'm sure others will gave the same question.

Nicely done by the way.  But there are a few changes I'll make to the menus
and other minor things.

1. Being slightly vision impaired I find the top banner - white text on a
gray background doesn't provide enough contrast for me to read the text.
2.  Menu options don't have enough line space between them if using your
finger instead of a stylus.
3.  A bolder font would be very beneficial for all.
4.  Switching from portrait to landscape display mode during functions such
as selecting an html file to use in Evil Portal  is kind of a PIA.

But it is functional and thats the most important.  Thanks for all of your
work on porting Marauder to the CYD..  Now I have something worthwhile to
use my CYDs for!