Fr4nkFletcher/ESP32-Marauder-Cheap-Yellow-Display #13: How to add battery inside the case?

DRCRecoveryData, 2024-03-12:
Hi,

I have successfully uploaded everything, but I'm curious about how to add a battery and an EEPROM blank slot on a PC for upgrading or adding an EEPROM chip. Can you provide guidance on this?

Fr4nkFletcher, 2024-03-12:
I wired a battery to the VIN/GND pins on the board side of the 4-pin connector at the bottom of the CYD. 

Check out the [CYD discord](https://discord.gg/nnezpvq) for more info/guidance on EEPROM mods

DRCRecoveryData, 2024-03-12:
> I wired a battery to the VIN/GND pins on the board side of the 4-pin connector at the bottom of the CYD. 
> 
> Check out the [CYD discord](https://discord.gg/nnezpvq) for more info/guidance on EEPROM mods

Does vin ( vcc ) is 5v or 3.3v?

Fr4nkFletcher, 2024-03-12:
5v comes out of the voltage regulator that VIN runs to when powered by USB. I used a PS3 controller battery at 3.7v and it works.

I found this also. I hope it helps https://github.com/hexeguitar/ESP32_TFT_PIO?tab=readme-ov-file#3-adding-psram

DRCRecoveryData, 2024-03-12:
> 5v comes out of the voltage regulator that VIN runs to when powered by USB. I used a PS3 controller battery at 3.7v and it works.
> 
> I found this also. I hope it helps https://github.com/hexeguitar/ESP32_TFT_PIO?tab=readme-ov-file#3-adding-psram 

[image]

How can i connect this gps and battery to 4 pin near micro usb?

Fr4nkFletcher, 2024-03-12:
> 
> How can i connect this gps and battery to 4 pin near micro usb?
> 
Connect the GPS using the 4pin connector. A cable for this usually ships with the CYD. I soldered 2 female dupont jumpers to GND and VIN at the board side of the 4-pin connector. On my ps3 controller lipo I clipped the 2 pin female jst connector and added male dupont jumpers. Plug and play to power off easily. I could've done this cleaner but was just trying to "make it work"

Fr4nkFletcher, 2024-03-12:
> 
> [image]
> 
> I've print the case

Can I buy a case from you?

DRCRecoveryData, 2024-03-12:
> 
> > 
> > [image]
> > 
> > I've print the case
> 
> Can I buy a case from you?

You can order via jlcpcb:
https://drive.google.com/drive/folders/1L7p5DK15z0mjWqkGOaK8IW8phT9H3rf_?usp=sharing

[image]

[image]

[image]

DRCRecoveryData, 2024-03-12:
> Connect the GPS using the 4pin connector. A cable for this usually ships with the CYD. I soldered 2 female dupont jumpers to GND and VIN at the board side of the 4-pin connector. On my ps3 [...]

Can you take picture of wirring? As i understand the gps vcc rx tx gnd connect same vcc and gnd to battery via a switch?

Fr4nkFletcher, 2024-03-12:
I checked out the print files and it looks like there's no opening for most of the buttons/connectors of the cyd. It is cool though

Fr4nkFletcher, 2024-03-12:
> Can you take picture of wirring? As i understand the gps vcc rx tx gnd connect same vcc and gnd to battery via a switch?

[image]

[image]

Fr4nkFletcher, 2024-03-12:
Apparently I didn't clip the jst connector on the battery. Pictures for reference

closed 2024-03-12 by Fr4nkFletcher

DRCRecoveryData, 2024-03-14:
[image]

[image]

The button maybe 15mm or 16mm

Fr4nkFletcher, 2024-03-14:
I like it

DRCRecoveryData, 2024-03-15:
The button i confirm is 15mm fit to hole, the screw is M3 12mm lenght, also does screen 2.4 inch working with firmware?

[image]

DRCRecoveryData, 2024-03-15:
How to connect gps module on your board?