Fr4nkFletcher/ESP32-Marauder-Cheap-Yellow-Display #14: CLI command not working

Pascal-m, 2024-03-17:
hello,
I can't place the cli commands on the CYD serial port, although I have the startup information.
Firmware version 0.13.9
Thk

Fr4nkFletcher, 2024-03-17:
> hello, I can't place the cli commands on the CYD serial port, although I have the startup information. Firmware version 0.13.9 Thk

This is because GPS in mapped to RX/TX (gpio 1,3) I could make another option in the webflasher for w/ or w/o gps if you need. Let me know

Pascal-m, 2024-03-17:
> > hello, I can't place the cli commands on the CYD serial port, although I have the startup information. Firmware version 0.13.9 Thk
> 
> This is because GPS in mapped to RX/TX (gpio 1,3) I could make another option in the webflasher for w/ or w/o gps if you need. Let me know

closed 2024-03-17 by Pascal-m

Fr4nkFletcher, 2024-03-17:
> ok, yes if you had the option in the webflasher that would be great!! THANKS

On it

Fr4nkFletcher, 2024-03-17:
> ok, yes if you had the option in the webflasher that would be great!! THANKS
> 
> > > hello, I can't place the cli commands on the CYD serial port, although I have the startup information. Firmware version 0.13.9 Thk
> > 
> > 
> > This is because GPS in mapped to RX/TX (gpio 1,3) I could make another option in the webflasher for w/ or w/o gps if you need. Let me know

Good to go

Pascal-m, 2024-03-17:
> > ok, yes if you had the option in the webflasher that would be great!! THANKS
> 
> On it

Well done !!
It works like a charme (y)
Thk :)