Fr4nkFletcher/ESP32-Marauder-Cheap-Yellow-Display #19: Want to see the battery percentage

Skilledcamman, 2024-04-25:
I want to be able to see battery percentage with tp4056

Skilledcamman, 2024-04-25:
Not sure how this could be done

Fr4nkFletcher, 2024-04-25:
> tp4056

Something like this might work https://www.aliexpress.us/item/3256805906634931.html?gatewayAdapt=glo2usa

More info can be found here on the ESP32Marauder wiki:
https://github.com/justcallmekoko/ESP32Marauder/wiki/battery

closed 2024-04-25 by Fr4nkFletcher

Skilledcamman, 2024-04-26:
> > tp4056
> 
> Something like this might work https://www.aliexpress.us/item/3256805906634931.html?gatewayAdapt=glo2usa
> 
> More info can be found here on the ESP32Marauder wiki: https://github.com/justcallmekoko/ESP32Marauder/wiki/battery

though you wouldnt be able to see it on the screen? also why does the boot configuring battery interface what is that?

Skilledcamman, 2024-04-26:
would I be able to implement it without i2c
and use tp4056 voltage and read it through io35