Fr4nkFletcher/ESP32-Marauder-Cheap-Yellow-Display #22: boot loop

Phat29072011, 2024-05-02:


Fr4nkFletcher, 2024-05-02:
Let me know

Phat29072011, 2024-05-03:
> Let me know

i use the cym web flasher and mine just Run to starting serial and it will restart

Fr4nkFletcher, 2024-05-03:
> > Let me know
> 
> i use the cym web flasher and mine just Run to starting serial and it will restart

Which module are you using?

Fr4nkFletcher, 2024-05-03:
> Serial monitor: 08:14:25.207 -> 08:14:25.207 -> Brownout detector was triggered 08:14:25.207 -> 08:14:25.207 -> ets Jul 29 2019 12:21:46 08:14:25.207 -> 08:14:25.207 -> rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT) 08:14:25.207 -> configsip: 0, SPIWP:0xee 08:14:25.254 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00 08:14:25.254 -> mode:DIO, clock div:1 08:14:25.254 -> load:0x3fff0030,len:1184 08:14:25.254 -> load:0x40078000,len:13260 08:14:25.254 -> load:0x40080400,len:3028 08:14:25.254 -> entry 0x400805e4 08:14:25.855 -> ESP-IDF version is: v4.4.5 08:14:26.045 -> Using TFT Shield 5 08:14:26.092 -> SPI_FREQUENCY: 40000000 08:14:26.092 -> SPI_READ_FREQUENCY:16000000 08:14:26.092 -> SPI_TOUCH_FREQUENCY: 2500000

Try a different cable

Phat29072011, 2024-05-03:
> > > Let me know
> > 
> > 
> > i use the cym web flasher and mine just Run to starting serial and it will restart
> 
> Which module are you using?

the cyd2usb on

Phat29072011, 2024-05-03:
> Try a different cable

it  can't too

Fr4nkFletcher, 2024-05-03:
> it can't too

If flashing with a new cable on a different port didn't help, I would try to compile from source and go from there. If that doesn't work, try with a different CYD/CYD2USB

closed 2024-05-05 by Fr4nkFletcher