Fr4nkFletcher/ESP32-Marauder-Cheap-Yellow-Display #25: Want analog battery circuit indicator.

Skilledcamman, 2024-05-25:
the esp32 marauder github repo has analog battery indicator developed by @Joshinken
https://github.com/justcallmekoko/ESP32Marauder/pull/35
https://github.com/justcallmekoko/ESP32Marauder/pull/31
https://github.com/justcallmekoko/ESP32Marauder/pull/30
I want to be able to do that on cyd marauder.

Joshinken, 2024-05-28:
oh hey thats me!
im skimming over it, doesnt the cyd use the marauder code? shouldnt it be merged in automatically?
hardware wise my method only needs 2 mosfets and a few resistors, though if youre not bothered by a small amount of background battery drain you can skip the mosfets

Joshinken, 2024-05-28:
im looking at it more closely now, yeah hardware wise this is doable, though id need some wrangling to make it not interfere with the boot process. using the power/uart header, have IO0/tx as the battery enable, IO2/rx as the battery measurement, 5v as 5v and gnd as gnd. youd need a secondary board containing charge and boost circuitry but all that is fairly easy

if youre willing to wait a few weeks i could get these parts and probably prototype a board for this tbh

Skilledcamman, 2024-05-28:
I thought you only needed 1 pin, I've seen some esp8266 battery indicator examples on [youtube](https://www.youtube.com/watch?v=CUUHYvZoduk) which only use one gpio pin, they use a voltage divider and connect the voltage to the battery
using io35 pin which only supports input according the  [cyd-repo](https://github.com/witnessmenow/ESP32-Cheap-Yellow-Display/blob/main/PINS.md)
This would free up the tx and rx pins which are usually used for gps for wardriving
also the secondary board your referring to is something like a tp4056?
also take your time, not like I'm paying you.

Joshinken, 2024-05-28:
you only need 1 pin to measure the battery, you need 2 pins if you dont want the battery to be drained 24/7 by the voltage divider
i need the pin header with rx and tx to get access to the 5v line so i thought id use those pins, though i guess you could plug a usb in or have a passthrough for the uart
secondary board means something containing all the electronics? you cant plug a battery straight into an esp32, you need to make a voltage that is used by the board, in this case thats 5v. to make a reasonable battery pack for this board you need a charge circuit to get power into the battery, a step up converter to get 5v, a voltage divider, and if you want the battery to not get drained, 2 mosfets and some pullup resistors

Skilledcamman, 2024-05-28:
> <Fr4nkFletcher/ESP32-Marauder-Cheap-Yellow-Display/issues/25/2134991909@
> github.com>

A lot of this went over my head, I'll just wait for you to figure it out.

Joshinken, 2024-05-28:
ok: a voltage divider relies on power flowing. power flowing means power flowing OUT OF your battery. this is generally considered "not good", so we build a circuit that disabled the voltage divider while it isnt needed. this isnt strictly necessary, but its good practice and doesnt cost much to do

an esp32 wants 3.3v and any more will kill it. the CYD provides that, but needs 5v itself, and any less wont work. a battery is between 3.2 and 4.2v, depending on how full it is, so if you plug it straight into any part of this you get a choice of either blowing up the esp or underpowering the CYD. so we need to get it up to 5v first, give that to the CYD, and let it provide the 3.3v to the esp32
 
you also need something to charge it, because batteries are very particular about their voltages and will decide to catch fire if you, say, try to give it 5v straight over usb; and something protect it from being discharged to much, and protect it from having too much power drawn at once. luckily something like a 4056 will handle those last 3, but it wont do either of the first 2, so we need a board that has all of these parts

closed 2024-06-03 by Fr4nkFletcher