Fr4nkFletcher/ESP32-Marauder-Cheap-Yellow-Display #27: GPS

HACKERprise, 2024-05-28:
The GPS is taking a long time to connect and catch satellites, Sometimes I even reboot several times But it says it's connected.

Fr4nkFletcher, 2024-05-28:
Which antenna are you using? It can take time to get a lock. Helps if you go outside

stevedee78, 2024-05-29:
I have the same problem. I have installed the NEO-6M.

HACKERprise, 2024-05-29:
> Which antenna are you using? It can take time to get a lock. Helps if you go outside

That's exactly what happens at the moment with the ceramic one.

HACKERprise, 2024-05-29:
But I would like to change the antenna, what do you recommend, and the Wi-Fi one too?

Fr4nkFletcher, 2024-05-29:
I use the NEO6MV2 with this antenna without issue https://a.co/d/be2EhWE

As for wifi, I use an Alfa 9dBi antenna for most projects 
https://a.co/d/1KuWTSE

I hope this helps

Fr4nkFletcher, 2024-05-29:
Will work on this today to make it better somehow.

Fr4nkFletcher, 2024-05-29:
[image]

https://github.com/justcallmekoko/ESP32Marauder/issues/338

closed 2024-05-29 by Fr4nkFletcher