Fr4nkFletcher/ESP32-Marauder-Cheap-Yellow-Display #29: Flash Error - ESP32-2432028R

pvonkalb, 2024-06-02:
I seem to be stuck in a loop. Its probably my fault but any help would be appreciated! 

Connecting...
Connected successfully.
Try hard reset.
[Za.readLoop:1] Finished read loop
Error: Couldn't sync to ESP. Try resetting.
Connecting...
Error: Failed to execute 'open' on 'SerialPort': The port is already open.

I plug in, select connect, select my com port through the browser, it says try resetting, I press the reset button screen goes black on the device, and then fails because its already open. Am I just missing something here? I can try a manual run if needed.

Fr4nkFletcher, 2024-06-02:
Are you on linux?

pvonkalb, 2024-06-02:
Win11Home. Do I need linux?

Fr4nkFletcher, 2024-06-02:
No, but this is a common error on Linux. 

First thing I would try would be a different cable. New incognito window -> plug in the CYD -> hold boot, press reset, and then let go of boot, and then attempt to flash.

pvonkalb, 2024-06-02:
So tried two additional cables in both Chrome and Opera incognito windows. Followed the same path. New incog window > plug in CYD > hold boot, press reset and then let go of boot > attempt to flash. 

Same error.  To add, this is a fresh CYD from Ali.

Fr4nkFletcher, 2024-06-03:
Can you try flashing this and let me know?
https://witnessmenow.github.io/Spotify-Diy-Thing/

pvonkalb, 2024-06-03:
This initially failed but it mentioned pressing the boot button while attempting the flash. That seemed to work! Thank you for your response and help with this!! Much appreciated.

Fr4nkFletcher, 2024-06-03:
Awesome!

closed 2024-06-03 by Fr4nkFletcher