Fr4nkFletcher/ESP32-Marauder-Cheap-Yellow-Display #30: No WPS support ?

OldMateAidos, 2024-06-12:
Would be awesome to see any form of WPS brute force/pixie attacks implemented

Fr4nkFletcher, 2024-06-12:
> Would be awesome to see any form of WPS brute force/pixie attacks implemented

100% agree it would be nice. It's theoretically possible but wouldn't be very practical I think due to the hardware just not being enough for this use case.

OldMateAidos, 2024-06-12:
Yh fair enough , I'm looking at trying to port over WPS features I'll fork and push if I manage it gets it working :)

Fr4nkFletcher, 2024-06-12:
> Yh fair enough , I'm looking at trying to port over WPS features I'll fork and push if I manage it gets it working :)

Sounds good to me!

closed 2024-06-12 by Fr4nkFletcher

OldMateAidos, 2024-06-12:
https://github.com/espressif/esp-idf hopefully I can start here has WPS support