Fr4nkFletcher/ESP32-Marauder-Cheap-Yellow-Display #33: ESP32-2432S024 not work display

Asmodeux2, 2024-06-26:
Will there be support for 2.4 inch screens or how can I make the screen work?

Fr4nkFletcher, 2024-06-29:
I'll have mine ready soon but this fork should get you there if you can't wait 

https://github.com/lsdlsd88/ESP32-Marauder-Cheap-Yellow-Display

closed 2024-06-30 by Fr4nkFletcher

Asmodeux2, 2024-07-22:
Fr4nkFletcher thanks for the tip. But no matter how much time I spend I can't get the project to compile trivially because of various errors. I realized that I need to _change parameters in User_Setup.h from TFT_eSPI library_. But at the moment of compilation everything is very tight. Please make different variations of screens in your CYD-Flasher. Thanks before.

P.S I am very stupid

Fr4nkFletcher, 2025-03-16:
@Asmodeux2 msg me on discord:@0xbangbang if you have the 2.4 cap and can test a bin

Fr4nkFletcher, 2025-03-20:
@Asmodeux2 https://github.com/Fr4nkFletcher/ESP32-Marauder-Cheap-Yellow-Display/commit/13ced215b7e8f2e90074205df57b2b9ce2a4e932