Fr4nkFletcher/ESP32-Marauder-Cheap-Yellow-Display #35: Screen touch control inverts after closing the "Select APs" menu

josilher, 2024-06-29:
Pretty much what the title says. After scanning APs I go to select APs to choose one and when I exit the menu is when the touch control gets inverted, pressing on the bottom on the screen is interpreted by the CYD as  if I pressed at the top of the screen. Every press feels mirrored on the Y axis meanwhile the X axis is ok.

Fr4nkFletcher, 2024-06-29:
I just tested a few of mine without issue. Which hardware and bin did you flash?

josilher, 2024-06-29:
I flashed it on the web flasher and chose the 1.0 two USB no GPS option since my CYD is the micro USB and type C one and I don't have a GPS module connected. The store from where I bought it says it's an ESP32-2432S028R but if you tried with a couple of CYD and can't find the error maybe mine is a fake one? Idk honestly I just came to report the error as soon as I encountered.

Fr4nkFletcher, 2024-06-29:
Flash this and then flash marauder again 

https://witnessmenow.github.io/Spotify-Diy-Thing/

josilher, 2024-06-29:
Okay now it works like a charm, don't know why I had to flash the spotify thing but hey if it works I'm not going to complain. Thanks!

closed 2024-06-29 by josilher