Fr4nkFletcher/ESP32-Marauder-Cheap-Yellow-Display #37: GUITION JC2432W328 ESP-32 2,8" web flasher works but display is black

maxpill, 2024-08-16:
The JC2432W328 ESP-32 2.8" web flasher successfully uploads code, but the display remains black. This issue persists despite multiple attempts. Tried all of the possible Marauder versions in the web flasher.

### To Reproduce

1. Upload Marauder to the JC2432W328 ESP-32 board using web flasher.

### Expected Behavior

The display should show the Marauder instead of remaining black.

### Actual Behavior

The display stays black after uploading the code. Serial console shows Marauder initializing properly.

### Board

[JC2432W328 ESP-32](https://pl.aliexpress.com/item/1005006729707613.html?gatewayAdapt=glo2pol)

maxpill, 2024-08-26:
Got it running omg :) I'd be happy to help adding it to your tool.

[image]

[image]

Fr4nkFletcher, 2024-08-26:
If you want you could send the bins and I could add it to the flasher. Nice work man

maxpill, 2024-08-27:
> If you want you could send the bins and I could add it to the flasher. Nice work man

Sure, I will link binaries in this issue. However I will test the firmware first, because I've already noticed some minor bugs regarding led pinout etc.

> You can change it to black bg by going into tft-espi user_setup.h and changing #define TFT_INVERSION to on/off

Already did, thanks :)

maxpill, 2024-08-29:
After further investigation, it appears that this display utilizes I2C Capacitive touch rather than the typical resistive touch used by CYD. Keeping my fingers crossed for a successful port to capacitive touch.

zidanjulio, 2024-09-09:
can I use it with a 3.5″ screen? Can I flash it directly on a website with a 3.5 screen?
I tried it before on a 2.8 screen and it worked.

maxpill, 2024-09-09:
> can I use it with a 3.5″ screen? Can I flash it directly on a website with a 3.5 screen?
> 
> I tried it before on a 2.8 screen and it worked.

What display controller does your board use?

zidanjulio, 2024-09-09:
like this 
[image]

Shirtsy, 2024-09-09:
@maxpill I have the same model of CYD, what changes did you have to make to get it to work? I'm curious to tinker with it myself.

xxjeanxx80, 2024-09-10:
I have a problem that, although not very serious, is still not trivial. Currently, after successfully uploading to my CYD, it does not recognize my touch input.
CYD : ESP32-2432S024R.  Although my box says  ST7789, I can only upload based on ILI9341 in User_Setup.h
https://github.com/user-attachments/assets/7dd5f359-eef6-4f68-b6ab-5b3114ba05fa
![Screenshot 2024-09-10 173042](http
[image]
s://github.com/user-attachments/assets/9c526aa4-4938-4f44-86d1-13b211e8054d)

maxpill, 2024-09-10:
> @maxpill I have the same model of CYD, what changes did you have to make to get it to work? I'm curious to tinker with it myself.

I will create a fork of this repo on my GitHub, but recently I'm short on time. Basically, there are a few things you need to do to get it running:

### Setup and Compilation

1. Set up your Arduino IDE to be able to compile this repository. You can use these two repos as guides:
	* https://github.com/smoochiee/MARAUDER-FOR-CYD---CHEAP-YELLOW-DISPLAY
	* https://github.com/justcallmekoko/ESP32Marauder/wiki/installing-firmware-from-source

### Board Information

2. Collect info on your particular board model as I did. What you need to know is:
	* Display pinout (as shown in the documentation I attached in this issue)
	* Display controller
	* Touch type (capacitive/resistive)
	* Touch controller

### Flashing and Testing

3. You can probably flash this Marauder software directly on your board using @Fr4nkFletcher's webflasher, and then use a serial console tool to see if it's functional.

### Display Configuration

4. To get the display working, you have to set up `User_Setup.h` with the proper pinout. You will also need to modify the repository Marauder code to work with your display controller.

### Touch Configuration

5. To get touch working, it's a bit harder:
	* If you have resistive touch, chances are it will work out of the box.
	* If not, you have to adjust the `calData` values.
	* If you have capacitive touch, you better wait for my repo. You have to basically change all the touch-related code in this repo, as the capacitive touch is handled totally differently.

zidanjulio, 2024-09-11:
How? Can the board that I sent with a 3.5inch screen be flashed directly through the webflasher?

maxpill, 2024-09-11:
> How? Can the board that I sent with a 3.5inch screen be flashed directly through the webflasher?

You can give it a try, with high degree of certainty the Marauder will work, but display will not.

closed 2024-09-20 by Fr4nkFletcher

Fr4nkFletcher, 2024-10-22:
@maxpill any updates sir?

VNEvil, 2024-11-07:
Hi @maxpill 
Is your archive complete?

maxpill, 2024-11-13:
@Fr4nkFletcher  @VNEvil Hello again. It is partially done, however I am recently very short on time, and as this is my hobby project it is not my priority recently.

Fr4nkFletcher, 2025-03-15:
@maxpill https://github.com/Fr4nkFletcher/ESP32-Marauder-Cheap-Yellow-Display/commit/5dd31b471d798256c70c959492d714ddd7612fd4 send me a link to your board's tft_espi libs and i will make you a bin if you want

Fr4nkFletcher, 2025-03-23:
@maxpill msg me on discord:0xbangbang if you still want to get this going

DangerLarry10, 2026-01-16:
@Fr4nkFletcher @maxpill  any chance you guys actually did get this going?? I thought I was close but I've got Cap display