Fr4nkFletcher/ESP32-Marauder-Cheap-Yellow-Display #42: Cannot get touch functionality on ESP32-2432S024

xxjeanxx80, 2024-09-10:
I have a problem that, although not very serious, is still not trivial. Currently, after successfully uploading to my CYD, it does not recognize my touch input.
CYD : ESP32-2432S024R.  Although my box says  ST7789, I can only upload based on ILI9341 in User_Setup.h

https://github.com/user-attachments/assets/7dd5f359-eef6-4f68-b6ab-5b3114ba05fa

[image]

[image]

Fr4nkFletcher, 2024-09-13:
Check out https://github.com/Fr4nkFletcher/ESP32-Marauder-Cheap-Yellow-Display/issues/40

closed 2024-09-13 by Fr4nkFletcher

Fr4nkFletcher, 2024-10-22:
@xxjeanxx80 try now =]