Fr4nkFletcher/ESP32-Marauder-Cheap-Yellow-Display #44: Still can't compile in Arduino IDE

jf1452, 2024-09-14:
Managed to make progress from my first attempt, but getting some more challenging issues:

c:/users/jonfu/appdata/local/arduino15/packages/esp32/tools/esp-x32/2302/bin/../lib/gcc/xtensa-esp32-elf/12.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: C:\Users\jonfu\AppData\Local\Temp\arduino\sketches\7013FBB08D57F1C60BA566D3359C0ADE\sketch\objs.a(Buffer.cpp.o):C:\Software\Arduino\ESP32-Marauder-Cheap-Yellow-Display-master\esp32_marauder/lang_var.h:188: multiple definition of `text_table4'; C:\Users\jonfu\AppData\Local\Temp\arduino\sketches\7013FBB08D57F1C60BA566D3359C0ADE\sketch\objs.a(BatteryInterface.cpp.o):C:\Software\Arduino\ESP32-Marauder-Cheap-Yellow-Display-master\esp32_marauder/lang_var.h:188: first defined here
c:/users/jonfu/appdata/local/arduino15/packages/esp32/tools/esp-x32/2302/bin/../lib/gcc/xtensa-esp32-elf/12.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: C:\Users\jonfu\AppData\Local\Temp\arduino\sketches\7013FBB08D57F1C60BA566D3359C0ADE\sketch\objs.a(Buffer.cpp.o):C:\Software\Arduino\ESP32-Marauder-Cheap-Yellow-Display-master\esp32_marauder/lang_var.h:187: multiple definition of `text_table3'; C:\Users\jonfu\AppData\Local\Temp\arduino\sketches\7013FBB08D57F1C60BA566D3359C0ADE\sketch\objs.a(BatteryInterface.cpp.o):C:\Software\Arduino\ESP32-Marauder-Cheap-Yellow-Display-master\esp32_marauder/lang_var.h:187: first defined here
c:/users/jonfu/appdata/local/arduino15/packages/esp32/tools/esp-x32/2302/bin/../lib/gcc/xtensa-esp32-elf/12.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: C:\Users\jonfu\AppData\Local\Temp\arduino\sketches\7013FBB08D57F1C60BA566D3359C0ADE\sketch\objs.a(Buffer.cpp.o):C:\Software\Arduino\ESP32-Marauder-Cheap-Yellow-Display-master\esp32_marauder/lang_var.h:186: multiple definition of `text_table2'; C:\Users\jonfu\AppData\Local\Temp\arduino\sketches\7013FBB08D57F1C60BA566D3359C0ADE\sketch\objs.a(BatteryInterface.cpp.o):C:\Software\Arduino\ESP32-Marauder-Cheap-Yellow-Display-master\esp32_marauder/lang_var.h:186: first defined here
c:/users/jonfu/appdata/local/arduino15/packages/esp32/tools/esp-x32/2302/bin/../lib/gcc/xtensa-esp32-elf/12.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: C:\Users\jonfu\AppData\Local\Temp\arduino\sketches\7013FBB08D57F1C60BA566D3359C0ADE\sketch\objs.a(Buffer.cpp.o):C:\Software\Arduino\ESP32-Marauder-Cheap-Yellow-Display-master\esp32_marauder/lang_var.h:185: multiple definition of `text_table1'; C:\Users\jonfu\AppData\Local\Temp\arduino\sketches\7013FBB08D57F1C60BA566D3359C0ADE\sketch\objs.a(BatteryInterface.cpp.o):C:\Software\Arduino\ESP32-Marauder-Cheap-Yellow-Display-master\esp32_marauder/lang_var.h:185: first defined here
c:/users/jonfu/appdata/local/arduino15/packages/esp32/tools/esp-x32/2302/bin/../lib/gcc/xtensa-esp32-elf/12.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: C:\Users\jonfu\AppData\Local\Temp\arduino\sketches\7013FBB08D57F1C60BA566D3359C0ADE\sketch\objs.a(Buffer.cpp.o):C:\Software\Arduino\ESP32-Marauder-Cheap-Yellow-Display-master\esp32_marauder/lang_var.h:184: multiple definition of `text_table0'; C:\Users\jonfu\AppData\Local\Temp\arduino\sketches\7013FBB08D57F1C60BA566D3359C0ADE\sketch\objs.a(BatteryInterface.cpp.o):C:\Software\Arduino\ESP32-Marauder-Cheap-Yellow-Display-master\esp32_marauder/lang_var.h:184: first defined here
c:/users/jonfu/appdata/local/arduino15/packages/esp32/tools/esp-x32/2302/bin/../lib/gcc/xtensa-esp32-elf/12.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: C:\Users\jonfu\AppData\Local\Temp\arduino\sketches\7013FBB08D57F1C60BA566D3359C0ADE\sketch\objs.a(CommandLine.cpp.o):C:\Software\Arduino\ESP32-Marauder-Cheap-Yellow-Display-master\esp32_marauder/lang_var.h:188: multiple definition of `text_table4'; C:\Users\jonfu\AppData\Local\Temp\arduino\sketches\7013FBB08D57F1C60BA566D3359C0ADE\sketch\objs.a(BatteryInterface.cpp.o):C:\Software\Arduino\ESP32-Marauder-Cheap-Yellow-Display-master\esp32_marauder/lang_var.h:188: first defined here
c:/users/jonfu/appdata/local/arduino15/packages/esp32/tools/esp-x32/2302/bin/../lib/gcc/xtensa-esp32-elf/12.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: C:\Users\jonfu\AppData\Local\Temp\arduino\sketches\7013FBB08D57F1C60BA566D3359C0ADE\sketch\objs.a(CommandLine.cpp.o):C:\Software\Arduino\ESP32-Marauder-Cheap-Yellow-Display-master\esp32_marauder/lang_var.h:187: multiple definition of `text_table3'; C:\Users\jonfu\AppData\Local\Temp\arduino\sketches\7013FBB08D57F1C60BA566D3359C0ADE\sketch\objs.a(BatteryInterface.cpp.o):C:\Software\Arduino\ESP32-Marauder-Cheap-Yellow-Display-master\esp32_marauder/lang_var.h:187: first defined here
c:/users/jonfu/appdata/local/arduino15/packages/esp32/tools/esp-x32/2302/bin/../lib/gcc/xtensa-esp32-elf/12.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: C:\Users\jonfu\AppData\Local\Temp\arduino\sketches\7013FBB08D57F1C60BA566D3359C0ADE\sketch\objs.a(CommandLine.cpp.o):C:\Software\Arduino\ESP32-Marauder-Cheap-Yellow-Display-master\esp32_marauder/lang_var.h:186: multiple definition of `text_table2'; C:\Users\jonfu\AppData\Local\Temp\arduino\sketches\7013FBB08D57F1C60BA566D3359C0ADE\sketch\objs.a(BatteryInterface.cpp.o):C:\Software\Arduino\ESP32-Marauder-Cheap-Yellow-Display-master\esp32_marauder/lang_var.h:186: first defined here
c:/users/jonfu/appdata/local/arduino15/packages/esp32/tools/esp-x32/2302/bin/../lib/gcc/xtensa-esp32-elf/12.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: C:\Users\jonfu\AppData\Local\Temp\arduino\sketches\7013FBB08D57F1C60BA566D3359C0ADE\sketch\objs.a(CommandLine.cpp.o):C:\Software\Arduino\ESP32-Marauder-Cheap-Yellow-Display-master\esp32_marauder/lang_var.h:185: multiple definition of `text_table1'; C:\Users\jonfu\AppData\Local\Temp\arduino\sketches\7013FBB08D57F1C60BA566D3359C0ADE\sketch\objs.a(BatteryInterface.cpp.o):C:\Software\Arduino\ESP32-Marauder-Cheap-Yellow-Display-master\esp32_marauder/lang_var.h:185: first defined here
c:/users/jonfu/appdata/local/arduino15/packages/esp32/tools/esp-x32/2302/bin/../lib/gcc/xtensa-esp32-elf/12.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: C:\Users\jonfu\AppData\Local\Temp\arduino\sketches\7013FBB08D57F1C60BA566D3359C0ADE\sketch\objs.a(CommandLine.cpp.o):C:\Software\Arduino\ESP32-Marauder-Cheap-Yellow-Display-master\esp32_marauder/lang_var.h:184: multiple definition of `text_table0'; C:\Users\jonfu\AppData\Local\Temp\arduino\sketches\7013FBB08D57F1C60BA566D3359C0ADE\sketch\objs.a(BatteryInterface.cpp.o):C:\Software\Arduino\ESP32-Marauder-Cheap-Yellow-Display-master\esp32_marauder/lang_var.h:184: first defined here
c:/users/jonfu/appdata/local/arduino15/packages/esp32/tools/esp-x32/2302/bin/../lib/gcc/xtensa-esp32-elf/12.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: C:\Users\jonfu\AppData\Local\Temp\arduino\sketches\7013FBB08D57F1C60BA566D3359C0ADE\sketch\objs.a(Display.cpp.o):C:\Software\Arduino\ESP32-Marauder-Cheap-Yellow-Display-master\esp32_marauder/lang_var.h:188: multiple definition of `text_table4'; C:\Users\jonfu\AppData\Local\Temp\arduino\sketches\7013FBB08D57F1C60BA566D3359C0ADE\sketch\objs.a(BatteryInterface.cpp.o):C:\Software\Arduino\ESP32-Marauder-Cheap-Yellow-Display-master\esp32_marauder/lang_var.h:188: first defined here
c:/users/jonfu/appdata/local/arduino15/packages/esp32/tools/esp-x32/2302/bin/../lib/gcc/xtensa-esp32-elf/12.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: C:\Users\jonfu\AppData\Local\Temp\arduino\sketches\7013FBB08D57F1C60BA566D3359C0ADE\sketch\objs.a(Display.cpp.o):C:\Software\Arduino\ESP32-Marauder-Cheap-Yellow-Display-master\esp32_marauder/lang_var.h:187: multiple definition of `text_table3'; C:\Users\jonfu\AppData\Local\Temp\arduino\sketches\7013FBB08D57F1C60BA566D3359C0ADE\sketch\objs.a(BatteryInterface.cpp.o):C:\Software\Arduino\ESP32-Marauder-Cheap-Yellow-Display-master\esp32_marauder/lang_var.h:187: first defined here
c:/users/jonfu/appdata/local/arduino15/packages/esp32/tools/esp-x32/2302/bin/../lib/gcc/xtensa-esp32-elf/12.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: C:\Users\jonfu\AppData\Local\Temp\arduino\sketches\7013FBB08D57F1C60BA566D3359C0ADE\sketch\objs.a(Display.cpp.o):C:\Software\Arduino\ESP32-Marauder-Cheap-Yellow-Display-master\esp32_marauder/lang_var.h:186: multiple definition of `text_table2'; C:\Users\jonfu\AppData\Local\Temp\arduino\sketches\7013FBB08D57F1C60BA566D3359C0ADE\sketch\objs.a(BatteryInterface.cpp.o):C:\Software\Arduino\ESP32-Marauder-Cheap-Yellow-Display-master\esp32_marauder/lang_var.h:186: first defined here
[... cut ...]
c:/users/jonfu/appdata/local/arduino15/packages/esp32/tools/esp-x32/2302/bin/../lib/gcc/xtensa-esp32-elf/12.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: C:\Users\jonfu\AppData\Local\Temp\arduino\sketches\7013FBB08D57F1C60BA566D3359C0ADE\sketch\objs.a(MenuFunctions.cpp.o): in function `MenuFunctions::orientDisplay()':
C:\Software\Arduino\ESP32-Marauder-Cheap-Yellow-Display-master\esp32_marauder/MenuFunctions.cpp:1154: undefined reference to `TFT_eSPI::setTouch(unsigned short*)'
collect2.exe: error: ld returned 1 exit status

exit status 1

Compilation error: exit status 1

Fr4nkFletcher, 2024-09-14:
@jf1452 Have you updated platform.txt?

jf1452, 2024-09-15:
Thanks, your suggestion sent me on a journey of learning Arduino 😊. I found [smoochiee](https://github.com/smoochiee/MARAUDER-FOR-CYD---CHEAP-YELLOW-DISPLAY?tab=readme-ov-file)'s instructions, faster compilation (I had the latest board definition loaded, not 2.0.10. number of warnings might be less, but still getting an error prevents compilation:

c:/users/jonfu/appdata/local/arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/esp-2021r2-patch5-8.4.0/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: C:\Users\jonfu\AppData\Local\Temp\arduino\sketches\7013FBB08D57F1C60BA566D3359C0ADE\sketch\objs.a(Buffer.cpp.o):C:\Software\Arduino\ESP32-Marauder-Cheap-Yellow-Display-master\esp32_marauder/lang_var.h:188: multiple definition of `text_table4'; C:\Users\jonfu\AppData\Local\Temp\arduino\sketches\7013FBB08D57F1C60BA566D3359C0ADE\sketch\objs.a(BatteryInterface.cpp.o):C:\Software\Arduino\ESP32-Marauder-Cheap-Yellow-Display-master\esp32_marauder/lang_var.h:188: first defined here
c:/users/jonfu/appdata/local/arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/esp-2021r2-patch5-8.4.0/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: C:\Users\jonfu\AppData\Local\Temp\arduino\sketches\7013FBB08D57F1C60BA566D3359C0ADE\sketch\objs.a(Buffer.cpp.o):C:\Software\Arduino\ESP32-Marauder-Cheap-Yellow-Display-master\esp32_marauder/lang_var.h:187: multiple definition of `text_table3'; C:\Users\jonfu\AppData\Local\Temp\arduino\sketches\7013FBB08D57F1C60BA566D3359C0ADE\sketch\objs.a(BatteryInterface.cpp.o):C:\Software\Arduino\ESP32-Marauder-Cheap-Yellow-Display-master\esp32_marauder/lang_var.h:187: first defined here
c:/users/jonfu/appdata/local/arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/esp-2021r2-patch5-8.4.0/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: C:\Users\jonfu\AppData\Local\Temp\arduino\sketches\7013FBB08D57F1C60BA566D3359C0ADE\sketch\objs.a(Buffer.cpp.o):C:\Software\Arduino\ESP32-Marauder-Cheap-Yellow-Display-master\esp32_marauder/lang_var.h:186: multiple definition of `text_table2'; C:\Users\jonfu\AppData\Local\Temp\arduino\sketches\7013FBB08D57F1C60BA566D3359C0ADE\sketch\objs.a(BatteryInterface.cpp.o):C:\Software\Arduino\ESP32-Marauder-Cheap-Yellow-Display-master\esp32_marauder/lang_var.h:186: first defined here
c:/users/jonfu/appdata/local/arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/esp-2021r2-patch5-8.4.0/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: C:\Users\jonfu\AppData\Local\Temp\arduino\sketches\7013FBB08D57F1C60BA566D3359C0ADE\sketch\objs.a(Buffer.cpp.o):C:\Software\Arduino\ESP32-Marauder-Cheap-Yellow-Display-master\esp32_marauder/lang_var.h:185: multiple definition of `text_table1'; C:\Users\jonfu\AppData\Local\Temp\arduino\sketches\7013FBB08D57F1C60BA566D3359C0ADE\sketch\objs.a(BatteryInterface.cpp.o):C:\Software\Arduino\ESP32-Marauder-Cheap-Yellow-Display-master\esp32_marauder/lang_var.h:185: first defined here
c:/users/jonfu/appdata/local/arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/esp-2021r2-patch5-8.4.0/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: C:\Users\jonfu\AppData\Local\Temp\arduino\sketches\7013FBB08D57F1C60BA566D3359C0ADE\sketch\objs.a(Buffer.cpp.o):C:\Software\Arduino\ESP32-Marauder-Cheap-Yellow-Display-master\esp32_marauder/lang_var.h:184: multiple definition of `text_table0'; C:\Users\jonfu\AppData\Local\Temp\arduino\sketches\7013FBB08D57F1C60BA566D3359C0ADE\sketch\objs.a(BatteryInterface.cpp.o):C:\Software\Arduino\ESP32-Marauder-Cheap-Yellow-Display-master\esp32_marauder/lang_var.h:184: first defined here
c:/users/jonfu/appdata/local/arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/esp-2021r2-patch5-8.4.0/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: C:\Users\jonfu\AppData\Local\Temp\arduino\sketches\7013FBB08D57F1C60BA566D3359C0ADE\sketch\objs.a(CommandLine.cpp.o):C:\Software\Arduino\ESP32-Marauder-Cheap-Yellow-Display-master\esp32_marauder/lang_var.h:188: multiple definition of `text_table4'; C:\Users\jonfu\AppData\Local\Temp\arduino\sketches\7013FBB08D57F1C60BA566D3359C0ADE\sketch\objs.a(BatteryInterface.cpp.o):C:\Software\Arduino\ESP32-Marauder-Cheap-Yellow-Display-master\esp32_marauder/lang_var.h:188: first defined here
c:/users/jonfu/appdata/local/arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/esp-2021r2-patch5-8.4.0/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: C:\Users\jonfu\AppData\Local\Temp\arduino\sketches\7013FBB08D57F1C60BA566D3359C0ADE\sketch\objs.a(CommandLine.cpp.o):C:\Software\Arduino\ESP32-Marauder-Cheap-Yellow-Display-master\esp32_marauder/lang_var.h:187: multiple definition of `text_table3'; C:\Users\jonfu\AppData\Local\Temp\arduino\sketches\7013FBB08D57F1C60BA566D3359C0ADE\sketch\objs.a(BatteryInterface.cpp.o):C:\Software\Arduino\ESP32-Marauder-Cheap-Yellow-Display-master\esp32_marauder/lang_var.h:187: first defined here
c:/users/jonfu/appdata/local/arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/esp-2021r2-patch5-8.4.0/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: C:\Users\jonfu\AppData\Local\Temp\arduino\sketches\7013FBB08D57F1C60BA566D3359C0ADE\sketch\objs.a(CommandLine.cpp.o):C:\Software\Arduino\ESP32-Marauder-Cheap-Yellow-Display-master\esp32_marauder/lang_var.h:186: multiple definition of `text_table2'; C:\Users\jonfu\AppData\Local\Temp\arduino\sketches\7013FBB08D57F1 [... cut ...]
c:/users/jonfu/appdata/local/arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/esp-2021r2-patch5-8.4.0/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: C:\Users\jonfu\AppData\Local\Temp\arduino\sketches\7013FBB08D57F1C60BA566D3359C0ADE\sketch\objs.a(MenuFunctions.cpp.o): in function `MenuFunctions::orientDisplay()':
C:\Software\Arduino\ESP32-Marauder-Cheap-Yellow-Display-master\esp32_marauder/MenuFunctions.cpp:1154: undefined reference to `TFT_eSPI::setTouch(unsigned short*)'
collect2.exe: error: ld returned 1 exit status

exit status 1

Compilation error: exit status 1

Fr4nkFletcher, 2024-09-15:
@jf1452  navigate to the directory you cloned the marauder repo to and run:
```bash
git submodule update --init --recursive
```
Move all your current Arduino libraries to a new dir somewhere else. Grab all the libraries from /ESP32-Marauder-Cheap-Yellow-Display/libraries and move them to your now empty Arduino library directory.

Can you can link your platform.txt file just to make sure? The multiple definitions error has me thinking that's part of the problem

closed 2024-09-20 by Fr4nkFletcher