Fr4nkFletcher/ESP32-Marauder-Cheap-Yellow-Display #45: Evil portal size too large

domia-official, 2024-09-21:
hello i got a problem , when i launch evil portal it say : the given html is too large .The byte limit is 11400 , but i see that the limit is 20ko and my index.html file is 16.ko
[image]
[image]

Fr4nkFletcher, 2024-09-21:
@domia-official can you link or share your html?

domia-official, 2024-09-21:
[index.html.txt](https://github.com/user-attachments/files/17084699/index.html.txt)

domia-official, 2024-09-21:
its html i just put .txt bc of github

Fr4nkFletcher, 2024-09-21:
[image]

I renamed your file to test.html and tested on my cyd2usb. It works?

domia-official, 2024-09-21:
[aaa.html.txt](https://github.com/user-attachments/files/17084795/aaa.html.txt)
check this one

domia-official, 2024-09-21:
the aaa.html.txt is the real problem

domia-official, 2024-09-21:
the problem was the svg logo , its seem to be not the real size so i use this website https://svgomg.net/ and choose the level 1 of compression . And now its good

closed 2024-09-21 by domia-official