Fr4nkFletcher/ESP32-Marauder-Cheap-Yellow-Display #48: splash + background colour

Lgndwilloske, 2024-09-23:
How do i change the boot splash and background colour like the pictures provided?

Fr4nkFletcher, 2024-09-23:
Flash the v0.13.10 .bins

JustCallMeKoko changed this in the most recent update (1.0.0)

Fr4nkFletcher, 2024-09-25:
Check out https://github.com/witnessmenow/ESP-Web-Tools-Tutorial

Bins are [here](https://github.com/Fr4nkFletcher/ESP32-Marauder-Cheap-Yellow-Display/tree/master/bins)

Lgndwilloske, 2024-09-25:
ahh done it thanks, one more question, is it possible to pair 2 esp's togeather? you get like some gpio pins with the cyd and i was wondering if you can use 2 esp's and what would/could it do??

Fr4nkFletcher, 2024-09-26:
Sounds like fun, what are you going for?

Lgndwilloske, 2024-09-26:
oh and one last question, I'm planning on making a v2 of this, with a esp32-c6, so it supports 6ghz bands and bluetooth 5, (educational purposes of course) i was wondering if there is a version of this that includes the TFT_eSPI library? so i can flash it and use a 2.8 touch screen. I was thinking of this screen: https://www.amazon.co.uk/dp/B082QVKT7M?tag=makeradvisor-21&linkCode=ogi&th=1&psc=1 Would you mind taking a minuet to see if i could do it? it would be realy appreciated

Thanks!

Lgndwilloske, 2024-09-27:
I thought it would be pretty cool considering the CYD is limited to 2.4ghz, So if the firmware can support 6ghz and this display that would be pretty cool

closed 2024-09-28 by Fr4nkFletcher