Fr4nkFletcher/ESP32-Marauder-Cheap-Yellow-Display #52: Change background color.

Muratik2803, 2024-09-25:
hello, im new here and i watched a lot of videos about this cheap yellow display and i dont know why but i cant change displau background. Its different from a picture instead of black i got white background and inversions. Please help me.

Muratik2803
9/25/24

Fr4nkFletcher, 2024-09-25:
Go into the tft-espi library user_setup.h and find the part about color inversion.

Muratik2803, 2024-09-25:
Ty but I don’t know where it is, can you like make steps where to go.

Fr4nkFletcher, 2024-09-26:
https://github.com/Fr4nkFletcher/ESP32-Marauder-Cheap-Yellow-Display/blob/master/libraries/TFT_eSPI-master/User_Setup.h#L115

Muratik2803, 2024-09-26:
and the last question. Thank for your kidness and fast replying but where to put it? i dont have a micro sd translator. Maybe you can like put it in your web site(CYM WEBFLASHER) i tried to put it into terminal (#define TFT_INVERSION_OFF) and nothing happened. 
Thanks for respond
Muratik2803

Muratik2803, 2024-09-26:
And the last question. Thank for your kindness and fast replying but where to put it? i don't have a micro SD translator. Maybe you can like put it in your web site (CYM WEBFLASHER) i tried to put it into terminal (#define TFT_INVERSION_OFF) and nothing happened.
Thanks for respond
Muratik2803 :)

closed 2024-09-26 by Muratik2803

Fr4nkFletcher, 2024-09-26:
I was thinking you were compiling from source. What cheap yellow display are you using?

Muratik2803, 2024-09-26:
It’s esp32-2432s028. And I got dark mode, in flasher instead of CYD2USB W/O GPS I picked CYD microusb only and I got dark version but when I watched the YouTube video of Talking Sasquach, he picked the CYD2USB W/O GPS. 

[image]

Fr4nkFletcher, 2024-09-26:
That's because he has the model with type-c and micro-usb so they call it the CYD2USB

Muratik2803, 2024-09-26:
But which one should I use?