Fr4nkFletcher/ESP32-Marauder-Cheap-Yellow-Display #53: CYM Web Flasher- Connected, No Boards in Board Selection Dropdown

Hard2Borrow, 2024-09-25:
I had a CYM that I flashed using the online web flasher; it worked, but the screen broke on that board. I got 2 new ones- from the same source- exactly the same... all are the ESP32-24325028R variant(the 1 most linked to from the resources/ same model- though 2 other variants exist being ESP32-24325028 & ESP32-24325028C once of which has both the micro and C USB, the other only having the micro USB, & 1 variant only having the USB C connection). I can see/ select my module to connect to... clicking connect gives me the following on all the 3 I have(1 of them has already been flashed with the Marauder firmware, but the screen is cracked):

_Connecting...
Connected successfully.
Try hard reset.
Chip type ESP32
Connected to ESP32 @ 115200 bps
MAC Address: XX XX XX XX XX XX XX (redacted)
Uploading stub...
Running stub...
Stub is now running...
Detecting Flash Size
FlashId: 0x164068
Flash Manufacturer: 68
Flash Device: 4016

================================_

I have tried to erase the boards, any and all reset, reset/ boot, boot button sequences/ pushes/ timings... all attempts, when connected properly give the above, nothing different other than the mac address between the different boards.

The drop down, where once, on the 1 that I initially flashed, is supposed to be populated with a few options that we are supposed to select and then finish off the flashing process.... but, **no options**.... just says please select board..clicking on the drop down gives me nothing.

I am using the latest, updated Chrome browser version.

More info from the online web flasher where I also attempted to erase the module and reconnected:

_Current Version:  

v1.0.0

   Disconnect   Serial Terminal   Clear Text
Dark Mode

--- SELECT BOARD ---

   
 Program
  
 Erase
       
Error: Failed to execute 'requestPort' on 'Serial': No port selected by the user.
Connecting...
Connected successfully.
Try hard reset.
Chip type ESP32
Connected to ESP32 @ 115200 bps
MAC Address: XX
Uploading stub...
Running stub...
Stub is now running...
Detecting Flash Size
FlashId: 0x164068
Flash Manufacturer: 68
Flash Device: 4016
Auto-detected Flash size: 4MB
[Fa.readLoop:1] Finished read loop
Connecting...
Connected successfully.
Try hard reset.
Chip type ESP32
Connected to ESP32 @ 115200 bps
MAC Address: XX
Uploading stub...
Running stub...
Stub is now running...
Detecting Flash Size
FlashId: 0x164068
Flash Manufacturer: 68
Flash Device: 4016
Auto-detected Flash size: 4MB

!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!
!!!               CAUTION!!!           !!!
!!!   THIS WILL ERASE THE FIRMWARE ON  !!!
!!!    YOUR DEVICE! THIS CAN NOT BE    !!!
!!!                UNDONE!             !!!
!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!

Erasing flash memory. Please wait...
Finished. Took 4819ms to erase.

---> ERASING PROCESS COMPLETED!

Connecting...
Connected successfully.
Try hard reset.
Chip type ESP32
Connected to ESP32 @ 115200 bps
MAC Address: XX
Uploading stub...
Running stub...
Stub is now running...
Detecting Flash Size
FlashId: 0x164068
Flash Manufacturer: 68
Flash Device: 4016
Auto-detected Flash size: 4MB_

Fr4nkFletcher, 2024-09-26:
This is different. Can you flash any of [these](https://github.com/witnessmenow/ESP32-Cheap-Yellow-Display/blob/main/PROJECTS.md) with success?

Lgndwilloske, 2024-09-27:
make sure your USB cable is a data cable, you might need to buy one. Install drivers so windows can read the USB SERIAL UART, It will probably show up as
COM6
COM7
COM8
If not, Hold boot while connecting
Use Chrome
Try a different USB port.
If all else fails There is a plan B:
Go to> Bins (on the repo) Download V0.13.10 W/O GPS> Go to [The ESPRESSIF Flasher](https://www.espressif.com/en/support/download/other-tools) And download the tool> Extract using Winrar etc> Run as ADMINISTRATOR> Select ESP32 in the first dropdown menu> make sure it says "Development" on the Second Drop Down menu> Click Ok> Open the .bin file i told you to download by clicking the (...) button> make sure that it is highlighted green and there is a tick next to it> Next to the file you selected there should be an empty text box, write: 0x10000> select The COM at the bottom, Anything but COM1> Click start> wait for it to flash> Click RST on the Board.

If it fails you can try erasing the esp32 using the flasher, If the tool doesent detect the board then restart the tool as administrator again

:) Hope it Works! Remember, Educational Purposes ONLY and keep it ethical.

Lgndwilloske, 2024-09-27:
You should be able to close this case Fr4nk 👍

closed 2024-09-28 by Fr4nkFletcher

Fr4nkFletcher, 2024-09-28:
@Hard2Borrow try again and copy-paste what you see in the Chrome Dev Console log