Fr4nkFletcher/ESP32-Marauder-Cheap-Yellow-Display #55: Deauth Flood doesnt seem to work

efsane14010, 2024-09-27:
Deauth Flood :
packets/sec: 0 

can this be caused because I dont have a sdcard inserted in it ?

Lgndwilloske, 2024-09-27:
go to sniffers and do, Beacon sniffer (20secs), AP sniffer (20secs), station sniffer(20secs) on the station sniffer you should see a mac address and a name of the network. Write down the mac address, go to general> select station> the mac address you wrote down> wifi attacks> deauth targeted and it will do it. Or deauth flood you need to select a AP to flood it

Remember, educational Purposes ONLY

Lgndwilloske, 2024-09-27:
SD cards are only for saving stations, SSID's, and AP's and evil portal.

closed 2024-09-28 by Fr4nkFletcher

efsane14010, 2024-10-02:
> go to sniffers and do, Beacon sniffer (20secs), AP sniffer (20secs), station sniffer(20secs) on the station sniffer you should see a mac address and a name of the network. Write down the mac [...]

Tyssmmmm it works Im soo dumb!

Lgndwilloske, 2024-10-02:
Don't blame thinks on intelligence, nobody is unintelligible. Your welcome.