Fr4nkFletcher/ESP32-Marauder-Cheap-Yellow-Display #58: Will not load into the firmware

Jasurios, 2024-10-07:
I have an esp32 2432S024 device but in the flasher it is displayed as esp32 2432S028R

After the firmware, the LED flashes first blue green and then red when loading and shows a black screen

Fr4nkFletcher, 2024-10-07:
Everything works but the display on these models. If you want to still use it, you can use the serial terminal for the marauder CLI

Jasurios, 2024-10-07:
it’s a pity, but how can I make it work, maybe a driver or something else. I just live sooooooooo far from civilization and I’ve been waiting for it for 1 year and I’m not eager to order a new one for a bunch of money

Fr4nkFletcher, 2024-10-07:
I'd have ordered the module that works already but the 2.4" is in the works hopefully it won't be long

closed 2024-10-07 by Fr4nkFletcher

Jasurios, 2024-10-07:
🤚

Lgndwilloske, 2024-10-07:
Download drivers and if that doesent work then hold boot press test and then let go of boot until the screen is black