Fr4nkFletcher/ESP32-Marauder-Cheap-Yellow-Display #60: Pls add .bins for ESP32-2432S024R

Asmodeux2, 2024-10-15:
---

## 🚀 ESP32 Marauder for ESP32-2432S024R (Beta) 🧪

Hello, @Fr4nkFletcher 

Please add the `.bin` files from this repository to Web Flasher: [ESP32Marauder_ESP32-2432S024R_CYD](https://github.com/Asmodeux2/ESP32Marauder_ESP32-2432S024R_CYD). The files are located in the `esp32_marauder/build/esp32.esp32.d32_pro` directory.

Note that there are still some issues with the touch screen, but it will function and display images. This should reduce the number of questions regarding this module. 💡

Please indicate that this is a **Beta** version. 🧪 Also, mention that all functions are operational through the CLI, and that work is being done on the touch screen. 🔧

Thank you! 🙏

---

Fr4nkFletcher, 2024-10-16:
Coming soon, just waiting on the module so I can test myself before adding.

closed 2024-10-16 by Fr4nkFletcher

Fr4nkFletcher, 2024-10-20:
@Asmodeux2 I just calibrated and made all the changes for the 2.4" and tested without any touch issues. What were you running into? Adding to the flasher soon for you to test

Asmodeux2, 2024-10-21:
Hi, @Fr4nkFletcher  , I have already looked at and tested your version, the global problem for both me and you is the CTS820 sensor driver which is not initialized. I will look at yours now, but anyway thank you very much for your work

Fr4nkFletcher, 2024-10-21:
Ok cool I'll look into this

Fr4nkFletcher, 2024-10-21:
@Asmodeux2 Are you certain you're not working with the 2432S024C?

Fr4nkFletcher, 2024-10-24:
@Asmodeux2 Pretty certain the resistive touch models don't employ the CST280