Fr4nkFletcher/ESP32-Marauder-Cheap-Yellow-Display #61: MicroSD Problems

Zynth9999, 2024-10-16:
Hey, last time you told me to reset until its fixed, didnt help, SD and Touch share the same SPI bus.

Lgndwilloske, 2024-10-16:
mind linking the issue post?

Fr4nkFletcher, 2024-10-16:
@Zynth9999 link to the specific module you're using please. Try flashing a previous version of Marauder and let me know if anything changes

Zynth9999, 2024-10-17:
> @Zynth9999 link to the specific module you're using please. Try flashing a previous version of Marauder and let me know if anything changes

I cant give you the link to the one i bought but its called ESP32-2432S028
Has 2 usb ports

Flashed 0.13.X - Nothing
Flashed 1.X - Still nothing

Its a software issue

As said on the CYD repo
[image]

Zynth9999, 2024-10-17:
> mind linking the issue post?
> 

https://github.com/Fr4nkFletcher/ESP32-Marauder-Cheap-Yellow-Display/issues/47

Fr4nkFletcher, 2024-10-17:
If you're using either of the linked compatible modules, the SD works. Try a different sdcard or module and let me know

Zynth9999, 2024-10-17:
Kk will try when i get home

Zynth9999, 2024-10-18:
Nope, didnt help

Lgndwilloske, 2024-10-18:
have you tried flashing different firmware on the esp? I'm pretty sure there's a firmware that can mount and unmount an sd card

Zynth9999, 2024-10-18:
> have you tried flashing different firmware on the esp? I'm pretty sure there's a firmware that can mount and unmount an sd card 

Yes i did

Zynth9999, 2024-10-18:
> there shouldn't be an issue with the software, maybe check what sd card you're using and if its supported?

Ive tried every card i can, from my wii, phone, 2ds and a new never used card

Fr4nkFletcher, 2024-10-18:
Take pictures of your module

Lgndwilloske, 2024-10-18:
> > have you tried flashing different firmware on the esp? I'm pretty sure there's a firmware that can mount and unmount an sd card
> 
> Yes i did

and did it work?

Zynth9999, 2024-10-18:
Nope it did not

Lgndwilloske, 2024-10-18:
well if you tried a different firmware (different from the maruder firmware) and the sd card didnt work then that is a clear sign that its the ESP-32 not the firmware. I recomend you send us the link or a few images of the esp so we can assess it, thanks

Fr4nkFletcher, 2024-10-18:
Take pictures of your module and we will continue this

closed 2024-10-21 by Fr4nkFletcher

Zynth9999, 2024-10-23:
[image]

[image]

Fr4nkFletcher, 2024-10-23:
It could just be the picture but GND and MOSI appear to be shorted

[image]

Zynth9999, 2024-10-24:
Let me check, but on the chip itself or the sd reader

Zynth9999, 2024-10-24:
[image]

You were right

Zynth9999, 2024-10-24:
Yep, i cut the part whwre it was connected and it works, thanks!

Fr4nkFletcher, 2024-10-24:
I knew there was something else going on! Glad you got it bro

Lgndwilloske, 2024-10-24:
sorry for being absent on this case and others, have been busy this week👍