Fr4nkFletcher/ESP32-Marauder-Cheap-Yellow-Display #62: Request for ESP32-Marauder CYD 3.5"C port (ESP32-3248S035C)

Colonel-Dixie, 2024-10-21:
Hello Fr4nkFletcher,    

Thank you for your AMAZING port of the Marauder firmware to the CYD! I've been playing with it in CYD 2.8"R 2USB with GPS and I'm loving it!          

Few days ago, I acquired a CYD variant, 3.5" Capacitive touch ESP32-3248S035C and I'm loving this hardware variant as it's slightly bigger which means more room in the case for battery, charging circuit, gps, etc. and the most important part that it's capacitive touch!    
  
I'm willing to be a tester if you want to test builds for you incase you don't have the hardware. I'm also willing to swap the esp module for the U32 version with the u.fl connector for better range, I already ordered one with 16MB from China for the 2.8" but if we got the 3.5 in time maybe I will swap the 3.5 Capacitive CYD first :D     
   
Thanks again for your awesome port and would love to see this port support more devices!

Fr4nkFletcher, 2024-10-21:
@Colonel-Dixie Thank you! I just ordered this — should arrive by Friday, and I'll have it by the weekend if all goes well. Also ordering the 3.2" soon

Fr4nkFletcher, 2024-10-24:
I ordered the resistive touch model to make it easy. Capactive will be later 

This will be here tomorrow
[image]

Colonel-Dixie, 2024-10-26:
@Fr4nkFletcher  Congratulations for having more boards to be supported! 

I think the one you have here is different from the one I have, I see it has a battery connector, that would be awesome if there is a 3.5C version of this board too!, even though I don't see a power button for it. I also see it has USB-C while the one I got has Micro-USB only.
    
      
 I have the more distributed version of 3.5C which is the same as 3.5R, like this one:   
[image]
[image]

>Capactive will be later
   
   Any ATA please? So I know if I should wait to install the 32U on the 3.5C or install it now on the 2.8R.   

I think the ultimate best CYD Marauder board will be the 3.5" capacitive for many reason like
1-Capastive touch! no need to push hard on screen to register input. 
2-Phone-ish screen size so it's bigger and easy to press.
3-LED is in front not the back blocked by the battery upgrade.
4-Bigger size means more room for battery, GPS, charging module, rocker switch etc..
5-Have light sensor above the screen for maybe future auto diming options.    
     
Thanks again and keep up the great work!

Fr4nkFletcher, 2025-03-16:
@Colonel-Dixie or anyone else with this board willing to test a bin message me on discord:@0xbangbang

Colonel-Dixie, 2025-03-17:
ESP32-Marauder-Cheap-Yellow-Display by Fr4nkFletcher for the ESP32-3248S035C is working great!      
I'm not home to test, but here is a video from a friend with the same ESP32-3248S035C board and it's working great! 
Thanks again @Fr4nkFletcher for the Amazing work and for not forgetting about this request!   
     
https://github.com/user-attachments/assets/5e95a452-e72c-432c-a257-e64c8c88548f

closed 2025-03-17 by Colonel-Dixie

Fr4nkFletcher, 2025-03-17:
Happy to help!