Fr4nkFletcher/ESP32-Marauder-Cheap-Yellow-Display #65: problem with touchscreen 2.4

Jasurios, 2024-10-27:
Hello, thank you for the new firmware, but when I flashed it, I realized that the screen does not work, plus the SD card is also not visible, what should I do?

Jasurios, 2024-10-27:
and by the way, the problem is not only with this but also with the color

Jasurios, 2024-10-27:
Help

Fr4nkFletcher, 2024-10-27:
@Jasurios u must have the capacitive version

Jasurios, 2024-10-27:
capacitive version what is this? Can you help me, I'm a newbie and I don't understand.

Jasurios, 2024-10-27:
[image]

Jasurios, 2024-10-27:
[image]

Fr4nkFletcher, 2024-10-27:
@Jasurios link to the module you purchased, please slow down though

Jasurios, 2024-10-27:
ESP32-2432S024

Jasurios, 2024-10-27:
I have discussed this topic before, thanks for adding the 2.4 screen option, but what about touch? For some reason it doesn't work for me.

Fr4nkFletcher, 2024-10-27:
Send me a link to what you have

Jasurios, 2024-10-27:
Смотри, что есть на AliExpress! Макетная плата ESP32 Arduino LVGL с Wi-Fi и Bluetooth, 2,4 дюйма, 240*320, умный экран дисплея 2,4 дюйма за 1 034,78 ₽
https://sl.aliexpress.ru/p?key=OnH1rbX

Jasurios, 2024-10-27:
This is where I bought the version capacitive touch

Jasurios, 2024-10-27:
[image]

Jasurios, 2024-10-27:
[image]

Jasurios, 2024-10-27:
I look forward to your reply.

Fr4nkFletcher, 2024-10-27:
You need the resistive touch version

Jasurios, 2024-10-27:
But you said it was a capacious version. Okay, I'll wait for updates😞

Jasurios, 2024-10-27:
You will make an update for this type of screen, right?

Jasurios, 2024-10-27:
Is there another way to control instead of touch, like a mouse and keyboard?

Fr4nkFletcher, 2024-10-27:
> But you said it was a capacious version. Okay, I'll wait for updates😞

No I was saying you must have the capacitive touch version which would be the reason it's not working for you. Only resistive right now.

Fr4nkFletcher, 2024-10-27:
> Is there another way to control instead of touch, like a mouse and keyboard?

Yes check out the serial terminal for marauder cli access

Fr4nkFletcher, 2024-10-27:
> You will make an update for this type of screen, right?

Yes I just need to order everything. Porting all resistive and the cap

Fr4nkFletcher, 2024-10-27:
> It's a pity, but will there be updates that will fix this?

In time yes

Fr4nkFletcher, 2024-10-27:
> It's a pity, but will there be updates that will fix this?

Order the modules that work for now?

closed 2024-10-27 by Fr4nkFletcher

Jasurios, 2024-10-27:
It's embarrassing to ask, but could you help me with loading sketches with my device? Yesterday I installed fresh libraries but I couldn't find on the Internet how to use them for my board.

Lgndwilloske, 2024-10-27:
an issue you have is your getting this of ali express, it's chinese, your better of ordering this stuff of amazon.co.uk the sketches thing, you can't use the ardunio ide with that esp it has to be an ardunio esp32, you will have to write the code and save it then flash it with a espressif flash tool.

Fr4nkFletcher, 2024-10-27:
Give me some time I'm not at home

Jasurios, 2024-10-27:
👍

Lgndwilloske, 2024-10-27:
> Give me some time I'm not at home

see, this is what i mean, we are busy people and we most likely won't reply so it's best if you wait a day or 2 if we don't reply, if you want you can always dm me on discord @lgndwilloski

Jasurios, 2024-10-27:
👍

Fr4nkFletcher, 2025-03-20:
@Jasurios added here https://github.com/Fr4nkFletcher/ESP32-Marauder-Cheap-Yellow-Display/commit/13ced215b7e8f2e90074205df57b2b9ce2a4e932