Fr4nkFletcher/ESP32-Marauder-Cheap-Yellow-Display #68: CYD support (2.4+Capactive)

Teapot321, 2024-11-01:
I would like you to add support for the new CYD.
I have this CYD and I can't figure out what it is.
It has a 2.4screen as well as a Capactive touch, ESP32-2432S024 is written on the back, this CYD has a Type-C port. It has pins (GND, O35, O22, O21) x2 as well as Bat+ Bat- and one I2C. The packaging said ESP32-2432S024C

Lgndwilloske, 2024-11-01:
i think it is being added

Fr4nkFletcher, 2024-11-01:
I don't have any of these here yet but it's in the works

closed 2024-11-01 by Fr4nkFletcher

Fr4nkFletcher, 2025-03-16:
@Teapot174 msg me on discord:@0xbangbang I have bin I needed tested for this board

Fr4nkFletcher, 2025-03-20:
Added here https://github.com/Fr4nkFletcher/ESP32-Marauder-Cheap-Yellow-Display/commit/13ced215b7e8f2e90074205df57b2b9ce2a4e932