Fr4nkFletcher/ESP32-Marauder-Cheap-Yellow-Display #75: AirTag spoofing file creator

TenoTrash, 2024-11-24:
Hello!
Is there a plan to include a way to create AirTag spoofing files? 
What kind of files is the AirTag spoofing looking for?
Thanks and Cheers!!!

Fr4nkFletcher, 2024-11-24:
Have you sniffed for Airtags first?

TenoTrash, 2024-11-24:
Sadly just one, but don't see any captured data in the microSD

Fr4nkFletcher, 2024-11-24:
Ouch, I haven't had any AirTags nearby to test so far. I'll get this sorted though

TenoTrash, 2024-11-24:
No worries, I just capture one, able to spoof, but it reboots and no data is saved. Trying to debug via serial, if found any will post here

Fr4nkFletcher, 2024-11-25:
> No worries, I just capture one, able to spoof, but it reboots and no data is saved. Trying to debug via serial, if found any will post here

Flash v1.1.0 and report back🫡

Fr4nkFletcher, 2024-11-25:
Thank you! Will keep working

Fr4nkFletcher, 2024-11-26:
@TenoTrash reflash and let me know plz

TenoTrash, 2024-11-26:
It do reset after working for about 5 minutes, can be heap ?
Will test it in the wild in the next few hours.
Cheers!

```
--------------------------------

         ESP32 Marauder      

            v1.1.0

       By: justcallmekoko

--------------------------------

> airtags: 0
Started BLE Scan
Guru Meditation Error: Core  0 panic'ed (LoadStoreError). Exception was unhandled.

Core  0 register dump:
PC      : 0x400e864d  PS      : 0x00060430  A0      : 0x8011cb0c  A1      : 0x3ffef1e0  
A2      : 0x40000000  A3      : 0x3ffef774  A4      : 0x3ffef7a4  A5      : 0x00000003  
A6      : 0x3ffef7a4  A7      : 0x0000004c  A8      : 0x0001085c  A9      : 0xffffffff  
A10     : 0x0000001e  A11     : 0x00000012  A12     : 0x00000019  A13     : 0x000000ff  
A14     : 0x00000000  A15     : 0x00004356  SAR     : 0x0000000c  EXCCAUSE: 0x00000003  
EXCVADDR: 0x40000000  LBEG    : 0x40092301  LEND    : 0x40092311  LCOUNT  : 0xffffffff  

Backtrace: 0x400e864a:0x3ffef1e0 0x4011cb09:0x3ffef2b0 0x40120f66:0x3ffef2f0 0x40125ad5:0x3ffef360 0x40125981:0x3ffef3a0 0x40125d9d:0x3ffef3c0 0x40124609:0x3ffef3e0 0x40081486:0x3ffef400 0x4011c122:0x3ffef420

ELF file SHA256: 21ff2c24f907b869

Rebooting...

```

Fr4nkFletcher, 2024-11-27:
Ok give it another try

TenoTrash, 2024-11-27:
Done, incognito mode flashed

```
#reboot
Rebooting...
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1184
load:0x40078000,len:13260
load:0x40080400,len:3028
entry 0x400805e4
ESP-IDF version is: v4.4.5
Evil Portal Found 2 HTML files
CLI Ready

              @@@@@@                        
              @@@@@@@@                      
              @@@@@@@@@@@                   
             @@@@@@  @@@@@@                 
          @@@@@@@      @@@@@@@              
        @@@@@@            @@@@@@            
     @@@@@@@                @@@@@@@         
   @@@@@@                      @@@@@@       
@@@@@@@              @@@@@@@@@@@@@@@@       
@@@@@                 @@@@@@@@@@@@@@@       
@@@@@                   @@@@@@@             
@@@@@                      @@@@@@           
@@@@@@                       @@@@@@@        
  @@@@@@                        @@@@@@@@@@@@
    @@@@@@@                          @@@@@@ 
       @@@@@@                     @@@@@@    
         @@@@@@@                @@@@@@      
            @@@@@@           @@@@@@         
              @@@@@@@      @@@@@@           
                 @@@@@@ @@@@@@              
                   @@@@@@@@@                
                      @@@@@@                
                        @@@@                

--------------------------------

         ESP32 Marauder      

            v1.1.0

       By: justcallmekoko

--------------------------------

> airtags: 0
Started BLE Scan
Guru Meditation Error: Core  0 panic'ed (LoadStoreError). Exception was unhandled.

Core  0 register dump:
PC      : 0x400e86f1  PS      : 0x00060230  A0      : 0x8011cba8  A1      : 0x3ffef4d0  
A2      : 0x40000000  A3      : 0x3ffef910  A4      : 0x3ffef940  A5      : 0x00000003  
A6      : 0x3ffef940  A7      : 0x0000004c  A8      : 0x000106c0  A9      : 0xffffffff  
A10     : 0x0000001e  A11     : 0x00000012  A12     : 0x00000019  A13     : 0x000000ff  
A14     : 0x00000000  A15     : 0x00004356  SAR     : 0x0000000c  EXCCAUSE: 0x00000003  
EXCVADDR: 0x40000000  LBEG    : 0x400922c9  LEND    : 0x400922d9  LCOUNT  : 0xffffffff  

Backtrace: 0x400e86ee:0x3ffef4d0 0x4011cba5:0x3ffef5a0 0x40120fd2:0x3ffef5e0 0x40125a49:0x3ffef650 0x401258f5:0x3ffef690 0x40125d11:0x3ffef6b0 0x40124615:0x3ffef6d0 0x4012ba06:0x3ffef6f0 0x4011c193:0x3ffef710

ELF file SHA256: 0f2b05047ee171bd

Rebooting...
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1184
load:0x40078000,len:13260
load:0x40080400,len:3028
entry 0x400805e4
ESP-IDF version is: v4.4.5
Evil Portal Found 2 HTML files
CLI Ready
```

Fr4nkFletcher, 2024-11-27:
... ok no worries will work harder

TenoTrash, 2024-11-27:
Well, just flash it onto a regular ESP32, run `sniffbt -t airtag` but have the very same result, please let me know if you need any help!

Fr4nkFletcher, 2024-11-28:
> Done, incognito mode flashed
>
> ```
> #reboot
> Rebooting...
> ets Jul 29 2019 12:21:46
>
> rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
> configsip: 0, SPIWP:0xee [...]

Try once more

TenoTrash, 2024-11-28:
> <Fr4nkFletcher/ESP32-Marauder-Cheap-Yellow-Display/issues/75/2505144987@
> github.com>

Flashed, will test in the wild today!

TenoTrash, 2024-11-28:
I don't have enought thanks! Will do further testing, but seems to work in perfect order!
Maybe a way so save the payload to the microSD, but this is outstanding work!
Cheers !!!

Fr4nkFletcher, 2024-11-28:
> I don't have enought thanks! Will do further testing, but seems to work in perfect order!
> Maybe a way so save the payload to the microSD, but this is outstanding work!
> Cheers !!!

This is good to see! Let me know if you find anything else, thank you for testing.

Fr4nkFletcher, 2024-11-29:
> Maybe a way so save the payload to the microSD
Flash one last time plz

Fr4nkFletcher, 2024-11-29:
> Maybe a way so save the payload to the microSD

@TenoTrash Flash one last time plz

TenoTrash, 2024-11-29:
> <Fr4nkFletcher/ESP32-Marauder-Cheap-Yellow-Display/issues/75/2507110789@

Done!
Using incognito mode, Chrome Linux Mint, CYD w/o GPS Latest 1.1.0

Will test it today in the wild, will reply this very email to post any
advance!

Thanks!

Fr4nkFletcher, 2024-11-29:
> Done! Using incognito mode, Chrome Linux Mint, CYD w/o GPS Latest 1.1.0 Will test it today in the wild, will reply this very email to post any advance! Thanks! El vie, 29 nov 2024 a la(s) 2:24 a.m., Fr4nkFletcher ( ***@***.***) escribió:
> […](#)
> -- Marcelo Ferrarotti "No me da miedo una IA que pase el test de Turing, me da más miedo que una IA lo desapruebe a propósito" <https://yip.su/16yWn7>

had to revert the log file changes, I found an issue when saving files. Shouldn't be long will hopefully be done tonight

TenoTrash, 2024-11-29:
Hello there! Just tested in the wild...
Sniffing and spoofing works nice (I'm able to detect from two ESP32 and one
CYM, and viceversa, they see and spoof as spected) but no data is saved on
the microSD.
Wich filename is supposed to create?

TenoTrash, 2024-11-29:
> <Fr4nkFletcher/ESP32-Marauder-Cheap-Yellow-Display/issues/75/2508165932@

Sorry, I just replied to the previous message, as I say, is stable but not
data is saved. Cheers!

Fr4nkFletcher, 2024-11-29:
> Sorry, I just replied to the previous message, as I say, is stable but not data is saved. Cheers! El vie, 29 nov 2024 a la(s) 2:06 p.m., Fr4nkFletcher ( ***@***.***) escribió:
> […](#)
> -- Marcelo Ferrarotti "No me da miedo una IA que pase el test de Turing, me da más miedo que una IA lo desapruebe a propósito" <https://yip.su/16yWn7>

@TenoTrash Flash once more and let me know if you have any issues saving files to the SD

TenoTrash, 2024-11-29:
> <Fr4nkFletcher/ESP32-Marauder-Cheap-Yellow-Display/issues/75/2508218204@

The .bin files that the webflasher use are the same from the bin direcory
in the repo?

Fr4nkFletcher, 2024-11-29:
I haven't updated the repo's bin but it's the 1128 build u want to flash. Flash incognito I just updated. Out and about right now

TenoTrash, 2024-12-01:
> @TenoTrash <https://github.com/TenoTrash> The webflasher repo has the
> up-to-date bins. Will add to the marauder repo later but if you flash
> current version right now you should see AirTag/flipper scans logging to
> the SD. I pulled the update at first bc of issues I've now concluded were
> due to my sd card. Let me know!

It works!!!

Fr4nkFletcher, 2024-12-03:
> It works!!!
> 
> 
> 
> El vie, 29 nov 2024 a la(s) 4:41 p.m., Fr4nkFletcher (
> 
> 
> 
> 
> > @TenoTrash <https://github.com/TenoTrash> The webflasher repo has the
> 
> > up-to-date bins. Will add to the marauder repo later but if you flash
> 
> > current version right now you should see AirTag/flipper scans logging to
> 
> > the SD. I pulled the update at first bc of issues I've now concluded were
> 
> > due to my sd card. Let me know!
> 
> >
> 
> 
> 
> 
> 
> 
> 
> 
> > <Fr4nkFletcher/ESP32-Marauder-Cheap-Yellow-Display/issues/75/2508279680@
> 
> > github.com>
> 
> >
> 
> 
> 
> 
> 
> 
> Marcelo Ferrarotti
> 
> "No me da miedo una IA que pase el test de Turing, me da más miedo que una
> 
> IA lo desapruebe a propósito" <https://yip.su/16yWn7>

Let me know if you need or find anything else!