Fr4nkFletcher/ESP32-Marauder-Cheap-Yellow-Display #78: airtag sniff crash

zapir69a, 2024-11-26:
when i try to do airtag sniff it's just reboots by it self everything else is working properly i tryed a few boards all of them had this problem   

                   -board: cyd 2 usb    
                   -version :1.10  without gps

Fr4nkFletcher, 2024-11-26:
On it

closed 2024-11-26 by Fr4nkFletcher

Fr4nkFletcher, 2024-11-27:
you wouldn't connect over serial and paste the stack trace when it crashes, would you?

Fr4nkFletcher, 2024-11-28:
> when i try to do airtag sniff it's just reboots by it self everything else is working properly i tryed a few boards all of them had this problem
> 
> ```
>                -board: cyd 2 usb    
>                -version :1.10  without gps
> ```

flash again and let me know.