Fr4nkFletcher/ESP32-Marauder-Cheap-Yellow-Display #84: CYD 2USB

CYD-USER, 2025-01-03:
The colors are inverted thats all

Fr4nkFletcher, 2025-01-03:
Which hardware exactly, and what version did you flash? I just tried this last night after making some updates to the webflasher and the colors were not inverted.

TenoTrash, 2025-01-03:
Do you flash using M5Stick-Launcher ?

Fr4nkFletcher, 2025-01-03:
@TenoTrash i do not

TenoTrash, 2025-01-03:
I've found that using M5Stick-Launcher and the .bin file from the release, colors are inverted, but I assume is not related since using the web flasher it just works as expected

stevedee78, 2025-01-03:
This Problem is only with the M5launcher.
I tried it.

Fr4nkFletcher, 2025-01-04:
> I've found that using M5Stick-Launcher and the .bin file from the release, colors are inverted, but I assume is not related since using the web flasher it just works as expected

Will fix

Fr4nkFletcher, 2025-01-04:
@CYD-USER I just updated the release bins. should be good

mxrxtr, 2025-01-04:
@Fr4nkFletcher this is my other acount gonna try it today thanks

CYD-USER, 2025-01-04:
@Fr4nkFletcher its fixed thanks

closed 2025-01-04 by CYD-USER