Fr4nkFletcher/ESP32-Marauder-Cheap-Yellow-Display #85: Color inversed on ESP32-2432S028R

nechry, 2025-01-09:
Hello, thank you for this project. I ordered and this module [ESP32-2432S028R](https://s.click.aliexpress.com/e/_onhDa2d)
However, the colors displayed are inverted. 
I used the web flasher with CYD2USB Latest 1.2.0

Is there a simple solution to correct this?

nechry, 2025-01-09:
quick update, for fun, I tried other module configurations
with:
- CYD MicroUSB-only w/o GPS, colors are good!
- CYD Type-C only, no display at all (normal it's the right display)

what is diff (other than usb connector) between CYD MicroUSB-only and CYD2USB ? just to know if I missed something else.

Fr4nkFletcher, 2025-01-09:
This is probably not the solution but reload your cache at the webflasher (CTRL+SHIFT+R)

Fr4nkFletcher, 2025-01-10:
Just updated, reload your cache and let me know.

closed 2025-01-10 by Fr4nkFletcher

nechry, 2025-01-10:
hello @Fr4nkFletcher I reload and also try in private session and perform webflasher, but same with CYD2USB image, the color is inverted.
with CYD MicroUSB-only it's ok. But as you said the only diff is the invert flag I will use with my device this config and it's ok

Fr4nkFletcher, 2025-01-12:
@nechry I'm not sure unless a change was made at the supply-chain level for these boards recently. I, along with others, have tested every variation with issue. I wish I had something more