Fr4nkFletcher/ESP32-Marauder-Cheap-Yellow-Display #88: White background

Neo2a, 2025-01-14:
I has been installed firmware to ESP32-2432S024R display but background white always. 
Touch fuction working but does not accurate. I have an Type3 board.

Fr4nkFletcher, 2025-01-14:
Link to your hardware please

Neo2a, 2025-01-15:
https://www.aliexpress.com/item/1005005865107357.html

Neo2a, 2025-01-15:
Touch screen chip is XPT2046

Fr4nkFletcher, 2025-01-15:
Will look into this evening thank you

Fr4nkFletcher, 2025-01-16:
@Neo2a take a picture of the back

Neo2a, 2025-01-16:
[image]

Component side of CYD

Fr4nkFletcher, 2025-01-16:
That's not the module linked in the repo. Order this https://a.co/d/fksX7NJ

closed 2025-01-16 by Fr4nkFletcher

Fr4nkFletcher, 2025-02-15:
@Neo2a should update the webflasher with this board later today

Fr4nkFletcher, 2025-02-16:
@Neo2a try now