Fr4nkFletcher/ESP32-Marauder-Cheap-Yellow-Display #93: Problem with Web Flashing

E-Olsen, 2025-01-19:
I am trying to flash an ESP32-2432S028 and this is the result:

Connecting...
Connected successfully.
Try hard reset.
[Ca.readLoop:1] Finished read loop
Error: Couldn't sync to ESP. Try resetting.

I have reset the board by holding down the reset button, pressing and releasing the Boot button and then released the Reset button.

The screen does reset but the output in the web window is the same.

E-Olsen, 2025-01-19:
My bad, missed the 'hold the Boot button while clicking Connect'.

closed 2025-01-19 by E-Olsen

Fr4nkFletcher, 2025-01-23:
I [updated](https://github.com/Fr4nkFletcher/ESP32-Marauder-Cheap-Yellow-Display/commit/69aaa02d5fe6f1de54b0cae3755fef8f03a81ad0#diff-b335630551682c19a781afebcf4d07bf978fb1f8ac04c6bf87428ed5106870f5R49) the README. should've done it sooner honestly