Fr4nkFletcher/ESP32-Marauder-Cheap-Yellow-Display #94: Crashes when trying to access the SD Card

dmgdscty, 2025-01-30:
If I use WiFi  - Sniffers - Scan APs for more then 30 seconds the device hard reboots.

If I use Devices - Save/Load Files Save APs works, but Load APs will also hard reboot the system.

Fr4nkFletcher, 2025-01-31:
> If I use WiFi - Sniffers - Scan APs for more then 30 seconds the device hard reboots.

This happens when the cyd runs out of memory. Did the free RAM indicator go to 0B in the statusbar?

> If I use Devices - Save/Load Files Save APs works, but Load APs will also hard reboot the system.

Use the Delete SD Files option under Device. This won't remove your EP files but it should fix your issue

closed 2025-01-31 by Fr4nkFletcher

dmgdscty, 2025-01-31:
> This happens when the cyd runs out of memory. Did the free RAM indicator go to 0B in the statusbar?

The RAM indicator goes from 70000B, 22000B, 12000B, then crashes around 2000B. I timed it and it was around 27 seconds. Is this normal? Does Scan APs always use so much ram, and if so is there anyway to fix this or is it just the fault of the cyd?

> Use the Delete SD Files option under Device. This won't remove your EP files but it should fix your issue

This seemed to work! Thanks!