Fr4nkFletcher/ESP32-Marauder-Cheap-Yellow-Display #96: Esp32-2432s032

protogenavali, 2025-02-10:
Hi, recently i purchased the 3.2inch cyd is there any chance there will be drivers-marauder itself available for it?

Fr4nkFletcher, 2025-02-10:
Resistive or capacitive touch? Both will be available soon with resistive being first more than likely.

closed 2025-02-10 by Fr4nkFletcher

protogenavali, 2025-02-13:
Okey if this info might help you the 3.5inch firmware works alright just the colours are inverted and the video is flipped.

Fr4nkFletcher, 2025-02-16:
Dm on discord @0xbangbang

Fr4nkFletcher, 2025-02-16:
Added in this [commit](https://github.com/Fr4nkFletcher/ESP32-Marauder-Cheap-Yellow-Display/commit/e3c422d1a8b9176381d251470daf6990b57ea4cd). Thanks for providing the needed touch calData!

[image]