Fr4nkFletcher/ESP32-Marauder-Cheap-Yellow-Display #97: CYD and NRF24

The-Chilliest-Willy-Ever, 2025-02-13:
Hey frank, been a while since we spoke.

Had an idea and that was to add a nrf24 to a cyd but; i do not have the pinout for it. So I was wondering if you had/knew the pinout?

Thanks - Willy (Formaly known as Lgndwilloski)

Fr4nkFletcher, 2025-02-14:
Use an [SD Sniffer](https://a.co/d/gpSRPXR)

[image]

NRF24 | CYD
-- | --
GND | GND
CE | 22
SCK | 18
MISO | 19
MOSI | 23
VCC | S3 test pad
IRQ | NC

closed 2025-02-14 by Fr4nkFletcher

The-Chilliest-Willy-Ever, 2025-02-15:
I don't own a sd sniffer, is there anyway I can do it on the actual sd port?

stevedee78, 2025-02-15:
[image]

Quelle Discord

stevedee78, 2025-02-15:
oder

[image]