Fr4nkFletcher/ESP32-Marauder-Cheap-Yellow-Display #101: error loading ap_0.pcap

HoaxParagon, 2025-02-22:
the file exists but upon trying to load it, it says"Could not open /APs_0.log". Seems like the firmware is seeking a log file instead of the saved pcap.

Fr4nkFletcher, 2025-02-22:
Pcaps should be reviewed on a separate machine using Wireshark or tcpdump. After sniffing access points, go to Save/Load Files, save the APs, and then try loading them again. That should work.

HoaxParagon, 2025-02-22:
I have done this process twice now, scanned APs, saved file, loading error. The .log file is never created, only pcap files with names like APs_3.pcap. trying after having deleted the sd card from inside the CYD marauder reboots the device. The load menu also doesn't allow selection of files. I also noticed that the 'reboot' button is very close to the rest of the menu options, any way to put it lower?

HoaxParagon, 2025-02-22:
it works now for some reason. I serial'd in and it wrote the file finally.

closed 2025-02-22 by Fr4nkFletcher