Fr4nkFletcher/ESP32-Marauder-Cheap-Yellow-Display #102: AirTag Sniff crashes / device reboot

PDBeal, 2025-02-25:
Trying the AirTag Sniff with the CYD causes it to reboot after a few seconds.  This is on the CYD MicroUSB-only V1.2.1 firmware on the ESP32-2432S028R

Serial Console on the Web flasher tool reports this:
`ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1184
load:0x40078000,len:13260
load:0x40080400,len:3028
entry 0x400805e4
[    24][D][esp32-hal-cpu.c:244] setCpuFrequencyMhz(): PLL: 480 / 2 = 240 Mhz, APB: 80000000 Hz
ESP-IDF version is: v4.4.5
I NimBLEDevice: "BLE Host Task Started"
I NimBLEDevice: "NimBle host synced."
Shutting down BLE
D NimBLEAdvertising: ">> stop"
D NimBLEAdvertising: "<< stop"
D NimBLEScan: ">> stop()"
D NimBLEScan: "<< stop()"
[   687][E][esp32-hal-cpu.c:110] addApbChangeCallback(): duplicate func=0x4013da9c arg=0x3ffc0cd4
[   689][W][sd_diskio.cpp:174] sdCommand(): no token received
Evil Portal Found 1 HTML files
[   848][I][esp32-hal-i2c.c:75] i2cInit(): Initialising I2C Master: sda=22 scl=27 freq=100000
Checking for battery monitors...
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1184
load:0x40078000,len:13260
load:0x40080400,len:3028
entry 0x400805e4
[    24][D][esp32-hal-cpu.c:244] setCpuFrequencyMhz(): PLL: 480 / 2 = 240 Mhz, APB: 80000000 Hz
ESP-IDF version is: v4.4.5
I NimBLEDevice: "BLE Host Task Started"
I NimBLEDevice: "NimBle host synced."
Shutting down BLE
D NimBLEAdvertising: ">> stop"
D NimBLEAdvertising: "<< stop"
D NimBLEScan: ">> stop()"
D NimBLEScan: "<< stop()"
[   682][E][esp32-hal-cpu.c:110] addApbChangeCallback(): duplicate func=0x4013da9c arg=0x3ffc0cd4
[   684][W][sd_diskio.cpp:174] sdCommand(): no token received
Evil Portal Found 1 HTML files
[   843][I][esp32-hal-i2c.c:75] i2cInit(): Initialising I2C Master: sda=22 scl=27 freq=100000
Checking for battery monitors...
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1184
load:0x40078000,len:13260
load:0x40080400,len:3028
entry 0x400805e4
[    24][D][esp32-hal-cpu.c:244] setCpuFrequencyMhz(): PLL: 480 / 2 = 240 Mhz, APB: 80000000 Hz
ESP-IDF version is: v4.4.5
I NimBLEDevice: "BLE Host Task Started"
I NimBLEDevice: "NimBle host synced."
Shutting down BLE
D NimBLEAdvertising: ">> stop"
D NimBLEAdvertising: "<< stop"
D NimBLEScan: ">> stop()"
D NimBLEScan: "<< stop()"
[   681][E][esp32-hal-cpu.c:110] addApbChangeCallback(): duplicate func=0x4013da9c arg=0x3ffc0cd4
[   683][W][sd_diskio.cpp:174] sdCommand(): no token received
Evil Portal Found 1 HTML files
[   842][I][esp32-hal-i2c.c:75] i2cInit(): Initialising I2C Master: sda=22 scl=27 freq=100000
Checking for battery monitors...
`

I had the Serial Console online for 2-3 consecutive crashes.  Was going to try and compile firmware from source to see if it was fixed and just not included in the web flash tool, but Arduino IDE was not able to compile without throwing a bunch of errors which may be due to different version of the IDE or the libraries.

PDBeal, 2025-02-26:
Just a GPS module plugged into the TX RX port next to the microUSB port.

PDBeal, 2025-02-27:
I noticed the back side of my CYD display shows ESP32-2432S028 and not the ESP32-2432S028R version.  I also tried the CYD that I received from HackerBox #0099.  It also was missing the R at the end of the part number on the back, but it too crashes on AirTag Sniff.  Tried it with and without GPS and with and without SD card.  Both end the same way with a reboot.

I also found it crashes on Flipper Sniff.  The Bluetooth Sniff seems to work and doesn't crash and the Detect Car Skimmers also seems to run without crashing the device.

PDBeal, 2025-02-27:
1.2.1 is the only option in the Web Installer when I picked the board.  I had thought about trying an older version to see if all of it had the same issue.

I tried the no gps one too, but all had the same thing.  I did notice the no GPS didn't have the same options underneath Bluetooth though.

Fr4nkFletcher, 2025-02-28:
Take a well-lit picture of the rear of the board please

Fr4nkFletcher, 2025-02-28:
Flash the w/o GPS build and then listen over serial when running flipper/at sniff. Should narrow down further

Fr4nkFletcher, 2025-02-28:
nothing connected to the 4pin connector above the sd slot?

Fr4nkFletcher, 2025-02-28:
Anything connected to the 4-pin connector above the SD slot? How are you powering also?

Flash a build without GPS support. GPS-enabled builds occupy the UART (RX/TX) lines, hiding serial output.

Flash this firmware and let me know the results: [Firmware Link](https://github.com/Fr4nkFletcher/Adafruit_WebSerial_ESPTool/blob/main/resources/PREVIOUS/esp32_marauder_v1_2_1_20250208_cyd_nogps.bin).

Please also take a well-lit picture of the rear of the board.

Fr4nkFletcher, 2025-02-28:
Anything connected to the 4-pin connector above the SD slot? How are you powering also?

Flash [this](https://github.com/Fr4nkFletcher/Adafruit_WebSerial_ESPTool/blob/main/resources/PREVIOUS/esp32_marauder_v1_2_1_20250208_cyd_nogps.bin) firmware and copy your serial output for flipper/at sniff

Please also take a well-lit picture of the rear of the board.

PDBeal, 2025-02-28:
> Fr4nkFletcher

Nothing connected to either of the GPIOs.  Only the GPS connected to the TXRX port.

I flashed the firmware you linked, and it did not crash.  Although, I based the manual flash on the screenshot on your page so I only flashed the single .bin file starting at offset 10000.  I figured the board already had the boot loader and partitions from the previous firmware flash.

Output from serial port:
airtags: 0
Started BLE Scan
RSSI: -89 MAC: E6:51:9A:B8:B3:A3
Len: 31 Payload: 1E FF 4C 00 12 19 10 99 47 9D 5B 86 36 64 F3 5E F2 FB C4 62 88 9E DC 8C 09 89 BA E1 57 03 1D 

RSSI: -85 MAC: FB:56:AC:AE:50:73
Len: 31 Payload: 1E FF 4C 00 12 19 6A 94 09 1A C2 4C 76 B6 8F C4 CB 2F C6 8D 6B 9C EA 1A 52 2F 7C 7B F4 01 90 

Scan complete!
Found 0 devices
Flippers: 0
Started BLE Scan

Picture of board is attached to the reply email.  Wasn’t quite sure how to attach to GitHub.  If it doesn’t make it I’ll see how to actually upload it somewhere.

￼

Fr4nkFletcher, 2025-02-28:
Ok cool so that narrows it down to the 02/14 commit. Will solve this weekend

closed 2025-02-28 by Fr4nkFletcher

Fr4nkFletcher, 2025-03-02:
Open the [flasher](https://fr4nkfletcher.github.io/Adafruit_WebSerial_ESPTool/) in an incognito window and try again. Still crash?

PDBeal, 2025-03-02:
That version works and does not crash.  Looks like you figured out the issue.

Thanks!

> On Mar 1, 2025, at 10:24 PM, Fr4nkFletcher ***@***.***> wrote:
> 
> flasher <https://fr4nkfletcher.github.io/Adafruit_WebSerial_ESPTool/>

Fr4nkFletcher, 2025-03-02:
I compiled most of the bins with esp core debugging set to verbose by mistake. Good issue thank you. Let me know if you need anything else