Fr4nkFletcher/ESP32-Marauder-Cheap-Yellow-Display #103: Problem with 2.2 Inch CYD ESP32-2432S022 (Guition)

BGrimm404, 2025-03-01:
I can´t get the screen working with this CYD - CYD ESP32-2432S022 (Guition).
Could you add this CYD to the list so that I can build a mini Marauder?
This could Help but it´s for another Sketch.
- Initial support for Guition ESP32-2432S022C - bmorcelli /
[M5Stick-Launcher](https://github.com/bmorcelli/M5Stick-Launcher)

Thank you.

closed 2025-03-01 by Fr4nkFletcher

BGrimm404, 2025-03-01:
Ok, thanks for the answer. 👍 I´ll wait till the fix is released.

munirhussain2872001, 2025-03-04:
Thank you sir you are a great person.🥰😘😍

munirhussain2872001, 2025-03-19:
> In the works, but not working as of yet

please sir help me

Fr4nkFletcher, 2025-03-19:
@munirhussain2872001 The 2.4" cap arrived today. It shares the same CST820 touch driver as the 2.2". close

munirhussain2872001, 2025-03-20:
> [@munirhussain2872001](https://github.com/munirhussain2872001) The 2.4" cap arrived today. It shares the same CST820 touch driver as the 2.2". close

Thanks and guide me

Fr4nkFletcher, 2025-03-20:
@munirhussain2872001 add me on discord:0xbangbang

munirhussain2872001, 2025-03-21:
please acccept my request name is   Black Lion