Fr4nkFletcher/ESP32-Marauder-Cheap-Yellow-Display #104: please help me sir

munirhussain2872001, 2025-03-03:
This is my board info https://github.com/rzeldent/platformio-espressif32-sunton?tab=readme-ov-file#esp32-2432s022-nc Please write a program for this board because you are the best coder but I am not 🥺🥺🥺🥺🥺😥😥😥😥😥😥😥😥😥       plz plz plz    

                                                                                   CYD-2432S022C (Capacitive)

munirhussain2872001, 2025-03-03:
[image]

munirhussain2872001, 2025-03-03:
only its work 😥😥😥😥😫😫😫😫

Fr4nkFletcher, 2025-03-03:
It's in the works -- soon!

closed 2025-03-03 by Fr4nkFletcher

munirhussain2872001, 2025-03-04:
Thank you sir you are a great person.🥰😘😍

munirhussain2872001, 2025-03-04:
> It's in the works -- soon!

[ESP32-Marauder-Cheap-Yellow-Display](https://github.com/Fr4nkFletcher/ESP32-Marauder-Cheap-Yellow-Display) really works on my board CYD-2432S022C (Capacitive) fuetur?

Fr4nkFletcher, 2025-03-05:
@munirhussain2872001 message me on discord. 3.2" capacitive working but I need you to test the 2.2" @0xbangbanghttps://github.com/user-attachments/assets/dc63df0f-a649-4c5e-83bb-a661acf112fb

Fr4nkFletcher, 2025-03-05:
@munirhussain2872001 message me on discord. 3.2" capacitive working but I need you to test the 2.2" @0xbangbang on discord 

https://github.com/user-attachments/assets/952109bc-9f05-450f-bcab-969c8341525d

munirhussain2872001, 2025-03-07:
> i have sent you dm from discord. i want to test your build for 2432s022c

i have not found you discord

munirhussain2872001, 2025-03-07:
> message me on discord. 3.2" capacitive working but I need you to test the 2.2" @0xbangbang on discord

Sir, Discord was not there, but I have created it. I don’t know how to use it much. Can you tell me which file I should use to make it work?

munirhussain2872001, 2025-03-07:
really sorry for late ans  😥

munirhussain2872001, 2025-03-07:
[image]     i use this  not work please help me sir 😯😫

msahin, 2025-03-07:
> [image] i use this not work please help me sir 😯😫

sir please don't flood. If you keep behaving like this no one will help you.

Fr4nkFletcher, 2025-03-07:
@munirhussain2872001 capacitive touch builds aren't at the webflasher yet.

munirhussain2872001, 2025-03-07:
> [@munirhussain2872001](https://github.com/munirhussain2872001) capacitive touch builds aren't at the webflasher yet.

where are?

munirhussain2872001, 2025-03-07:
https://discord.gg/UUBXwdn3

munirhussain2872001, 2025-03-10:
Sir please I need you help 😭

munirhussain2872001, 2025-03-19:
sir, please help me 😥

Fr4nkFletcher, 2025-03-19:
sir

munirhussain2872001, 2025-03-20:
> sir

??? I not understand

munirhussain2872001, 2025-03-20:
> sir
Please guide me