Fr4nkFletcher/ESP32-Marauder-Cheap-Yellow-Display #106: ESP32-2432S032R Display Bugs

pipeninja, 2025-03-14:
I've been testing the release that supports ESP32-2432S032R and I'm noticing some bugs and general quality of life stuff that I think needs adjusting. I thought some feedback might be useful.

Firstly, transitioning between different screens there is a red full screen redraw...pretty sure this isn't necessary...not really a problem as it doesn't affect functionality, but it's kinda whack.

Secondly, on screens that have lists of output (like beacon scanning etc), the output doesn't scroll, new output is drawn over the top of what is already there. This is quite annoying and makes certain screens difficult to read and use.

Finally, it seems bmorcelli over at the M5 Launcher / Bruce project has figured out how to get the GT911 touch working on the ESP32-2432S032C...any chance that can be integrated into Marauder?

stevedee78, 2025-03-14:
`Secondly, on screens that have lists of output (like beacon scanning etc), the output doesn't scroll, new output is drawn over the top of what is already there. This is quite annoying and makes certain screens difficult to read and use.`

https://github.com/Fr4nkFletcher/ESP32-Marauder-Cheap-Yellow-Display/issues/105

Fr4nkFletcher, 2025-03-14:
> I've been testing the release that supports ESP32-2432S032R and I'm noticing some bugs and general quality of life stuff that I think needs adjusting. I thought some feedback might be useful. [...]

@pipeninja ESP32-2432S032C will be up later this evening hopefully. Almost done

closed 2025-03-14 by Fr4nkFletcher

Fr4nkFletcher, 2025-03-15:
@pipeninja https://github.com/Fr4nkFletcher/ESP32-Marauder-Cheap-Yellow-Display/commit/5dd31b471d798256c70c959492d714ddd7612fd4

Fr4nkFletcher, 2025-03-23:
> Secondly, on screens that have lists of output (like beacon scanning etc), the output doesn't scroll, new output is drawn over the top of what is already there. This is quite annoying and makes certain screens difficult to read and use.

Problem found. message me on discord:0xbangbang if you want the updated bin before i add it to the webflasher

Fr4nkFletcher, 2025-03-23:
> Firstly, transitioning between different screens there is a red full screen redraw...pretty sure this isn't necessary...not really a problem as it doesn't affect functionality, but it's kinda whack.
> 
On it

Fr4nkFletcher, 2025-03-29:
@pipeninja all fixed in https://github.com/Fr4nkFletcher/ESP32-Marauder-Cheap-Yellow-Display/releases/tag/v1.3.1

pipeninja, 2025-04-06:
Nice. Apols for the delayed response! I'll give it a go shortly.