Fr4nkFletcher/ESP32-Marauder-Cheap-Yellow-Display #108: Touch points are shifted/misplaced in landscape mode on CYD35 board!

Madani64, 2025-03-20:
I am running the revision 1.2.1 on a CYD35 board. Everything is working fine except that when I go to the "Select AP" screen--which is in landscape orientation-- the "X Back" will not respond to touch! I discovered that if I touch the mirror side of the screen where the "X Back" is shown, it works and I can exit the page. 
I checked the code but couldn't work out how or if I can modify it to correct the issue. I'd appreciate help and advice. 

Also, is there a way to increase the font sizes generally for the entire application? Fonts look to small for a 3.5 inch display.

Thank you!

Mark.

Fr4nkFletcher, 2025-03-20:
Force reload at the webflasher. Should be good to go

closed 2025-03-20 by Fr4nkFletcher

Madani64, 2025-03-20:
It is a resistive touch. Force reloaded via webflasher to the same result!

Fr4nkFletcher, 2025-03-20:
can you take a pic of the back of the board or link to your hardware? I just tested mine without issue

Madani64, 2025-03-21:
[image]

Fr4nkFletcher, 2025-03-21:
These are cool. I'll have to order as that's not the 3.5" resistive board linked under [compatibility](https://github.com/Fr4nkFletcher/ESP32-Marauder-Cheap-Yellow-Display?tab=readme-ov-file#compatibility)

Madani64, 2025-03-22:
Thank you Fr4nk :-) Will you consider sharing the correctives to solve the screen issues please?

Fr4nkFletcher, 2025-03-22:
Can you link to that board or any examples on GitHub?

Fr4nkFletcher, 2025-03-22:
@Madani64 msg me on discord:0xbangbang

Madani64, 2025-03-22:
https://www.aliexpress.com/item/1005008394843311.html?spm=a2g0o.order_list.order_list_main.25.440e1802AAlVms