Fr4nkFletcher/ESP32-Marauder-Cheap-Yellow-Display #117: TFT_eSPI

Wiilly-i-am, 2025-04-14:
Hey Frank, bit off topic 

I'm making a project of my own, but I'm hitting a roadblock with TFT_eSPI. I'm Using a ESP32-2432S028, and I think the issue is down to my User_Setup.h 

Could you help me?

Here is my Config:
-----
```
// User setup for ESP32-2432S028
#define USER_SETUP_INFO "ESP32-2432S028 CYD Setup"

#define ILI9341_DRIVER

// Pin configuration for the CYD display
#define TFT_MISO 19
#define TFT_MOSI 23
#define TFT_SCLK 18
#define TFT_CS   5
#define TFT_DC   15
#define TFT_RST  4

// Backlight pin
#define TFT_BL   32
#define TFT_BACKLIGHT_ON HIGH

// Font loading
#define LOAD_GLCD
#define LOAD_FONT2
#define LOAD_FONT4
#define LOAD_FONT6
#define LOAD_FONT7
#define LOAD_FONT8
#define LOAD_GFXFF
#define SMOOTH_FONT

// SPI settings
#define SPI_FREQUENCY       40000000
#define SPI_READ_FREQUENCY  20000000
#define SPI_TOUCH_FREQUENCY 2500000

Wiilly-i-am, 2025-04-14:
Oh and in Arduinio IDE (Thats what im using) i have 3 Options:
* ESP32 Wrover Module
* ESP32 Wrover Kit (all Versions)
* upesy ESP32 Wrover DevKit

Fr4nkFletcher, 2025-04-14:
Check for your board here. This has everything needed for TFT_eSPI for all of the cheap yellow displays, in one place 

https://github.com/Fr4nkFletcher/ESP32-Marauder-Cheap-Yellow-Display/blob/master/libraries/TFT_eSPI-CYD

Wiilly-i-am, 2025-04-14:
how would i know which board is which?

Fr4nkFletcher, 2025-04-14:
Link to your hardware please

Wiilly-i-am, 2025-04-14:
Pretty sure its this one: https://www.amazon.co.uk/ESP32-2432S028-Development-Touchscreen-Bluetooth-Compatible/dp/B0D2D6HTDR It has the same Model Number "ESP32-2432S028

Fr4nkFletcher, 2025-04-14:
Should be here then https://github.com/Fr4nkFletcher/ESP32-Marauder-Cheap-Yellow-Display/blob/master/libraries/TFT_eSPI-CYD/2.8CYD

Wiilly-i-am, 2025-04-14:
do i just replace all the files??

Fr4nkFletcher, 2025-04-14:
Yes sir

Fr4nkFletcher, 2025-04-14:
Add me on discord:0xbangbang if you ever need help with anything I got you

Wiilly-i-am, 2025-04-17:
is there anything else we can talk on apart from discord?

Wiilly-i-am, 2025-04-17:
also im making huge progress on it all