Fr4nkFletcher/ESP32-Marauder-Cheap-Yellow-Display #118: touch inverted

SamuelPerez401, 2025-04-14:
Hello, I hope everybody here is ok. I'm having an issue when using my CYD board. I use the web flasher and choose the right type I have, but when I'm trying to use it, I see that the direction of the touch is inverted. I mean, the text appears to be on the left side of the display, but if I want to select an option I have to touch where the text is but on the right side. When the screen turns horizontally, if the texts are on top, I have to touch on the bottom. Can someone help me out please

Fr4nkFletcher, 2025-04-14:
@SamuelPerez401 No worries, can you link to the board you have?

SamuelPerez401, 2025-04-14:
yeah, sure. I have a 3.5 inch resistive touch screen 320x480 esp32-32E link: https://share.temu.com/GFQoGcmptnC

Fr4nkFletcher, 2025-04-15:
I've seen these pop up recently. Reach out on discord and can make it work I'm sure

closed 2025-04-15 by Fr4nkFletcher

Fr4nkFletcher, 2025-04-19:
Your board isn't supported per the compatible devices list. I don't have this device in hand but if you want to send a PR for this board I'm all for it