Fr4nkFletcher/ESP32-Marauder-Cheap-Yellow-Display #120: RESISTANCE TOUCH NOT WORKING ON MODEL: 3248S035R

WeakBlood, 2025-04-29:
Hello, the webflasher does not have a model: ESP32-3248S035R RES:320*480 MICRO USB Resistive, so i tried with the typce-c resistive but the touch is not responding at all even after  checking the resistive touchscreen with the original firmware an trying multiple versions is not working, so i think it's just a compatibility issue. Even if it is not working for me still thanks for the beautiful project <3

Fr4nkFletcher, 2025-04-30:
Can you snap a shot of the board back?

WeakBlood, 2025-04-30:
> Can you snap a shot of the board back?

[image]

Fr4nkFletcher, 2025-04-30:
This isn't the module supported unfortunately, check the compatible device links in the readme for what works

closed 2025-04-30 by Fr4nkFletcher