Fr4nkFletcher/ESP32-Marauder-Cheap-Yellow-Display #122: Touch not working on CYD ESP32 2432S028R

wrrrrn, 2025-05-07:
First, big thanks for this fantastic project! It's my first foray into electronics, and this has been an ideal start! 

I'm trying to build the project from source. It builds and flashes successfully, but touch stops working on the screen.

I have one of these https://www.amazon.com/dp/B0CLR7MQ91 and followed the smoochiee tutorial closely, but I'm a bit stumped. 

Any help/advice would be greatly appreciated!

Fr4nkFletcher, 2025-05-07:
Display works but not touch?

Fr4nkFletcher, 2025-05-07:
Use this library and clear arduino cache then upload once more

https://github.com/Fr4nkFletcher/ESP32-Marauder-Cheap-Yellow-Display/tree/master/libraries/TFT_eSPI-CYD/2.8CYD2USB