Fr4nkFletcher/ESP32-Marauder-Cheap-Yellow-Display #125: screen background is white

edwardtich1, 2025-05-18:
The screen background is white and the inscriptions are not visible

Board ES32-2432S028

Fr4nkFletcher, 2025-05-18:
How/what did you flash? Which board exactly?

edwardtich1, 2025-05-18:
[image]

Fr4nkFletcher, 2025-05-18:
Link to your board please

edwardtich1, 2025-05-18:
https://aliexpress.ru/item/1005006996177674.html?spm=a2g2w.orderdetail.0.0.29084aa6llFE4Y&sku_id=12000038988234537

edwardtich1, 2025-05-18:
The function needs to invert the display

Fr4nkFletcher, 2025-05-18:
New variants every day it seems like. This could work but I'd have to have it in hand to make it work as the listing you sent, in the description, links to the files for the 2.4". Also it says capacitive touch but in the listing image it shows an XPT2046, which is the resistive touch controller. Take pics of your board if you can please

Fr4nkFletcher, 2025-05-18:
Oh shit just flash single USB 2.8

edwardtich1, 2025-05-18:
[image]

Fr4nkFletcher, 2025-05-18:
If the CYD2USB options has inverted colors, flash the Single and it should be all good.

edwardtich1, 2025-05-18:
https://bruce.computer/flasher 
work is good

edwardtich1, 2025-05-18:
> If the CYD2USB option has inverted colors, flash the Single and it should be all good.

everything is fine
now on a black background

closed 2025-05-18 by Fr4nkFletcher