Fr4nkFletcher/ESP32-Marauder-Cheap-Yellow-Display #126: SD Card not working on "ESP32-32E 240x320 2.8" LCD resistive touch"

MisterFenek, 2025-05-18:
SD Card not working on "ESP32-32E 240x320 2.8" LCD resistive touch"
at boot the logs say unable to mount sd card, sd card not supported
using latest version bin

MisterFenek, 2025-05-18:
aint no way i just used all of my 5 sd cards and the one that wasnt working properly somehow worked on esp32

Fr4nkFletcher, 2025-05-18:
These boards are fairly new and may be a little different from the sunton boards linked under [compatibility](https://github.com/Fr4nkFletcher/ESP32-Marauder-Cheap-Yellow-Display?tab=readme-ov-file#compatibility). But according to the link you shared, the pinout for the sd slot should be the same.

[image]

[image]

I would try another sd less than 16gb in size if it doesn't mount after multiple reboots.

MisterFenek, 2025-05-18:
> These boards are fairly new and may be a little different from the sunton boards linked under [...]

the sd cards i tried were 1gb, 2gb, 4gb chinese, 8g, 8g chinese, only the last one worked somehow

MisterFenek, 2025-05-18:
also, is it possible to extract wps config? when scanning it shows "RDd WPS config of [network]" i think this means it got a wps config from AP or am i wrong?

MisterFenek, 2025-05-18:
ok