Fr4nkFletcher/ESP32-Marauder-Cheap-Yellow-Display #133: ADD SUPPORT/PORT for Lilygo T-HMI 2.8 RESISTIVE ESP32-S3

nocapppin, 2025-06-06:
hi, if you are able to look into porting over CYM to the Lilygo T – HMI 2.8 inch display that would be great due to the large amount of PS RAM and flash size. It would be great to run marauder and has the same specs as a regular CYD displaybesides having a more powerful ESP32-S3. if you're able to add it that would be amazing as I'm sure there are a ton of people out there that would love to use your CYM on there Lily go T – HMI thank you.

Fr4nkFletcher, 2025-06-06:
Should be fun. Will try to order this today

nocapppin, 2025-06-06:
Great sounds good. Looking forward to it 🤘

closed 2025-06-06 by Fr4nkFletcher

nocapppin, 2025-06-07:
Cool. Check Aliexpress or Alibaba got good price. And has so much potential to be the best CYD with 8mb PSRAM,16Mb Flash. Willl work wonderfully

nocapppin, 2025-06-07:
Can't wait till you get it in and port it💯💯😮‍💨😮‍💨

Fr4nkFletcher, 2025-06-07:
I wouldn't call this a CYD but https://a.co/d/8pqxdnF

nocapppin, 2025-06-07:
I mean it’s pretty cheap if you compare it to some rlly cheaper ones that are maybe $5 then it’s expensive but for what it has built in chipset and build quality is great you’ll see you once you get yours 👍🏻 you’ll love it

nocapppin, 2025-06-09:
Let me know if it has been ported and which ones to choose when flashing if it has been added, thank you. Thank you.